vvip-68/GyverPanelWiFi #222: GyverPanelWiFi

EDUARDT68, 2022-03-22:
Привет.
Я в перечислении фишек GuverPanelWiFi увидел вот это: 
**Новое!** _Синхронизация эффектов между устройствами. Может быть создано до 10 групп устройств, количество устройств в группе - не ограничено._
**Новое!** _Беспроводная трансляция эффектов на матрицу устройств из специализированных программ типа Jinx!, Wixen Lights, xLights и подобных по протоколу sACN Streaming ACN E1.31 DMX_
Можно об этом по подробнее? Где то есть описание?

vvip-68, 2022-03-22:
Да, конечно. Всё в wiki - целый раздел - "Синхронизация устройств".

closed 2022-03-22 by vvip-68

EDUARDT68, 2022-03-22:
Ага. Спс

EDUARDT68, 2022-03-22:
Тут ещё вопрос нарисовался. Я прошил обновление. Примерно минут пять всё работает нормально, а потом включаются ночные часы. Хотя бы примерно можете сказать что может быть? Куда копать?

vvip-68, 2022-03-22:
1. Настройки режимов по времени.
2. Возможно при неправильных настройках подключения кнопки (особенно если ее нет) или применения неправильных версий ядра / библиотек - срабатывает эффект, когда программа думает что кнопка нажата и уменьшает яркость до минимума. При достижении минимальной яркости, когда эффекты уже не могут отображаться - включается режим отображения часов.

EDUARDT68, 2022-03-22:
Хм!??? Уже 12 минут работает без проблем. Надеюсь что именно кнопка проблемой была.
Но пока не собираю в корпус. понаблюдаю.