vvip-68/GyverPanelWiFi #224: Управление через умный дом

bulava92, 2022-03-23:
Настраиваю работу с матрицей в NodeRED по mqtt, чтобы управлять через HomeKit с Siri и Алису. Сделал управление вкл/выкл, настройку яркости, добавил выбор нужных мне эффектов. Вопрос по настройке эффектов, в api управления устройством есть три параметра:
1. $8 1 N D;   - D -> параметр #1 для эффекта N;
2. $8 3 N D;   - D -> параметр #2 для эффекта N;
3. $8 6 N D;   - D -> контрастность эффекта N;

К примеру эффект часы, в приложении команда для настройки эффекта называется "Вариант", какой параметр из указанных выше надо отправлять? Если логически подумать, то первый параметр
Эффект лампа, в приложении три команды для настройки - контрастность, цвет и насыщенность, правильно понимаю что для управления надо использовать соответственно параметры из api в порядке 3, 2, 1?
А эффекта звездопад параметры в приложении - контрастность, скорость, вариант, также надо параметры из api в порядке 3, 2, 1?

Не совсем очевидно для меня. Было бы например проще один параметр в api для контрастности эффекта, один для скорости и т.д.

P.S. Могу позже прислать готовый код из Node RED для размещения в Wiki в интеграциях в умный дом. Лично для меня оказалось проще всё это настроить в Node RED - с домовенком я не сталкивался и не стал разбираться, в Home Assistant попробовал, но параметров много и тоже всё слишком сложно для меня, а вот в Node RED после того как разобрался с API устройства сделал всё за пару минут и доволен работой

vvip-68, 2022-03-23:
Да, согласен. Этот момент плохо документирован. А я за давностию лет уже плохо помню особенности реализации. Делалось так, потому что проект развивался постепенно - то одно добавлялось, то другое и порою - бессистемно. А скурпулёзно все задокументировать - нужно время и порой, больше чем на само написание и отдладку кода. Никто не любит писать документацию, особенно когда что-то делается "чисто для себя"...

В общем случае - "параметр 1" - это то, что в приложении в настройках обозначено как "Вариант". Используется почти для всех эффектов. 

"параметр 2" - только для некоторых эффектов, для которых более одной настройки (кроме скорость и контрастность). Например для эффекта "Узоры" -  "параметр 1" - первая строка настроек "Вариант" задает направление движения узора, а "параметр 2" - Вариант, расположенный ниже - какой именно узор выводится.  Для некоторых эффектов - "параметр 2" - это чекбокс "Сегменты".

Проще всего чтобы быть уверенным - смотреть в мониторе порта какие команды для эффектов отправляются при изменении настроек эффектов.

Если вы все-таки решитесь довести дело до конца - буду признателен, если в процессе изучения составите таблицу эффектов и что для каждого из них есть "параметр 1" и "параметр 2" - я размещу потом ее в wiki.

Также буду благодарен за ваш вариант реализации интеграции с Node RED - размещу ссылку на ваш проект в wiki/

>К примеру эффект часы, в приложении команда для настройки эффекта называется "Вариант", какой параметр из указанных выше надо отправлять? Если логически подумать, то первый параметр

Для часов - параметр 1 - цвет отображения часов $8 1 N D; параметр 2 - не используется

> Эффект лампа, в приложении три команды для настройки - контрастность, цвет и насыщенность, правильно понимаю что для управления надо использовать соответственно параметры из api в порядке 3, 2, 1?

Для Лампа - параметр 1 - цвет фона (самой "лампы") $8 1 N D, параметр 2 - "насыщенность" $8 3 N D
В каком порядке передавать команды - разницы нет, это отдельные команды. Впрочем, кажется, их можно было писать 'в одной строке', отправляемый в топик - вроде "$8 1 N D;$8 1 N D;" - но это нужно проверить.

> А эффекта звездопад параметры в приложении - контрастность, скорость, вариант, также надо параметры из api в порядке 3, 2, 1?

Для эффекта "Звездопад" 
Контрастность - это контрастность - $8 6 N D; - D -> контрастность эффекта N;
Скорость - это скорость - $15 V 0; - V - скорость 0..255
Вариант - параметр 1 - для звездопада - плотность звездопада - $8 1 N D; параметр 2 - не используется в этом эффекте

---

Изначально был канал управления из приложения через UDP, потом понадобилось чтобы была возможность управлять издалека, вне зоны действия локальной сети роутера. Был прикручен MQTT, а для упрощения кода - по каналу команд (топик cmd) - просто передавалась команда в том же самом формате, как она приходит из приложения - и парсится тем же самым куском кода. Поэтому вот так. На момент написания про интеграцию в умный дом даже не предполагалось (соответственно писалось без учета требований - их некому было выдвигать, а у меня умного дома нет), это потом пользовали как-то стали его прикручивать, приспосабливая то, что есть.

bulava92, 2022-03-23:
Спасибо за ответы
Когда закончу - пришлю ссылку вам на свои исходники к Node Red

bulava92, 2022-03-27:
[flows.json.zip](https://github.com/vvip-68/GyverPanelWiFi/files/8357579/flows.json.zip)
[image]

В файле мой конфиг для управления матрицей через Алису и Siri с помощью Node Red
Что сделано:
1. Управление наложением часов на эффект (вкл/выкл). Для того чтобы работала данная функция, надо чтобы она включена была в приложении. Если в приложении выключено - то при включении через HomeKit ничего не произойдет
2. Управление размером часов (большой/малый)
3. Управление яркостью матрицы
4. Возможность сделать полную заливку матрицы цветом через включение режима рисования. Сначала инициируется режим рисования, затем через пару секунд отправляется команда с заданным цветом
5. Через устройство "Телевизор" добавил нужные мне эффекты в хомкит. 
6. Возможность настроить у проброшенных эффектов в хомкит параметры, а именно контраст, вариант и скорость, параметры подгружаются с эффектом. Примеры - включен эффект "Лава", у него есть три параметра - скорость, контраст и вариант, значения отображаются в хомкит. Если потом включить эффект часы, то параметры скорость и контраст обнулятся, так как не поддерживаются данным эффектом, таким образом к выбранному эффекту подтягиваются установленные ранее значения.

Если опыт работы с Node Red есть то все настроится за пару минут, главное включить интеграцию с Хомкитом, Алисой и MQTT

vvip-68, 2022-03-27:
Спасибо. Но я в этом ничего не понимаю.
Будет ли инструкция как что делать для тех, кто захочет это повторить?
Или дать им ссылку (в wiki) на ваш конфиг, а дальше - пусть сами разбираются?

vvip-68, 2022-03-27:
Есть ли у вас репозиторий, где все это лежит, чтобы была всегда актуальная версия? Я бы в wiki на репозиторий ссылку выложил.

bulava92, 2022-03-27:
Опубликовал - https://github.com/bulava92/nodered_matrix

Инструкция для новичков будет, но немного позже, а тем кто разбирается в node red достаточно повторить то что я сделал, указав наименования своих аккаунтов

closed 2022-04-09 by vvip-68