vvip-68/GyverPanelWiFi #226: led corner blinking

dio1985, 2022-04-17:
[image]

i have 4panels(16×16)
i found 1point blinking led. so i tried to change to other panel. but its same
and also i changed led direction by code. but its same

any problem only for me?

vvip-68, 2022-04-17:
It is common issue for those, who do not want read instructions on several first lines of main sketch file and uses wrong version of ESP8266 core. It should be 2.7.4 version, not 3.x.x

vvip-68, 2022-04-17:
After you install right version of esp8266 core du not forget also return led bin back to D2 from D4, that uses in case of core 3.x.x version

closed 2022-04-17 by vvip-68