vvip-68/GyverPanelWiFi #231: auto connection wifi

dio1985, 2022-05-17:
"auto connect" has some problem when start

1. panel is connected xxx.xxx.xxx.100
2. phone is connected xxx.xxxx.xxx.101
3. The app is connected to addr xxx.xxx.xxx.100 
  (Set up an automatic connection checked)

but If i change the order of Wi-Fi connection after the connection is successful

1. phone is xxx.xxx.xxx.100 
2. (panel is not connected yet) now powe off
3. insert power . panel is xxx.xxx.xxx.101
4. start app. (app try to connect 100 that is phone)
5.  The phone is connected, but the application is successful. 
But I couldn't load the existing settings
I can't do anything

vvip-68, 2022-05-17:
Sorry, it is not clear description for issue. I do not understand completely where is trouble.
If you set in app specified IP address and checkbox 'Autoconnect' is checked, obviously app will try to connect to specified IP without necessity to tap on button 'Connect". What`s wrong?

dio1985, 2022-05-17:
> Sorry, it is not clear description for issue. I do not understand completely where is trouble. If you set in app specified IP address and checkbox 'Autoconnect' is checked, obviously app will try to connect to specified IP without necessity to tap on button 'Connect". What`s wrong?

For example, the ip of the panel is 100. The application connection is successful. After that, turn off all power connected to wifi again The panel is connected to ip 101 (in this case, the phone has ip 100). But the app connects to the phone with ip 100. But I can't do anything.

dio1985, 2022-05-17:
I mean, the app is connected to the IP of the phone.

vvip-68, 2022-05-17:
If you specify panel IP directly in WiFiPanelSetup application, panel couldn`t change IP in random manner, it uset always exactly specified IP. If you leave in WiFiPanelSetup IP address empty or set it to 0.0.0.0 - panel will use DHCP and will get IP address given by your router.

It is your choice how to setup panel. If you prefer use  DHCP - you may setup your router for set specified IP to device each time on panel request getting IP by DHCP. See manual for your router.

If you want, you may specify IP address for panel as static in WiFiPanelSetup application and panel will receive specified IP each time on start (on power on).

If you can`t connect to panel, when it connected to your local network, you may shortly press button on panel immediately after it is powered on. In this case panel will create Access Point with IP 192.168.4.1.  Connect your phone to this Access Point and use WiFiPanelSetup application for setup static IP.

closed 2022-05-17 by vvip-68