vvip-68/GyverPanelWiFi #232: Цвета в Jinx

KrasAnt, 2022-05-17:
Добрый день! Собрал матрицу. Все прекрасно работает, кроме отображения цвета в эффектах созданных в Jinx. В настройках Output Patch>Fast Patch менял всевозможные варианты последовательности цветов, но правильно все равно не показывает. Эффект Матрица показывает красным, а огонь зелёный.

vvip-68, 2022-05-17:
Если же у вас вывод эффектов Jinx не из файлов с SD-карты, а напрямую с Jinx через трансляцию - единственный способ все же разобраться с настройками цвета в самом Jinx - перекодирование тут не поможет. Удалите всё, связанное с предыдущей попыткой настроить и начните всё "с чистого листа", досконально следуя инструкции.

vvip-68, 2022-05-17:
Раз эффекты из скетча показываются правильно - какой смысл менять ее в скетче? Настраивайте Jinx "from scratch".

closed 2022-05-17 by vvip-68

KrasAnt, 2022-05-17:
Эффекты проигрываются с Sd-карты. Ладно, буду разбираться. Спасибо за ответ.