vvip-68/GyverPanelWiFi #233: getWeekdayString

dio1985, 2022-05-20:
file . utility.ino

 String getWeekdayString(uint8_t wd) {
  String str = "";
  switch (wd) {
    case  1: str = F("Monday");    break;
    case  2: str = F("Tuesday");   break;
    case  3: str = F("Wednesday"); break;
    case  4: str = F("Thursday");  break;
    case  5: str = F("Friday");    break;
    case  6: str = F("Saturday");  break;
    case  7: str = F("Sunday");    break;
  }  
  return str;
}

I just changed the day of the week to english but it doesn't show properly

"Fri2a22" ??????????

https://user-images.githubusercontent.com/78635592/169562382-9c6f0aff-e6b2-4678-90c9-187d3903f28b.mp4

vvip-68, 2022-05-20:
I have no answer. Video is inaccessible.

dio1985, 2022-05-20:
> I have no answer. Video is inaccessible.

https://user-images.githubusercontent.com/78635592/169616596-2c264d37-b3d1-491b-b124-32ea6ad8f69e.mp4

> I have no answer. Video is inaccessible.

https://user-images.githubusercontent.com/78635592/169616996-fcf7e99c-7807-4188-8469-c16c2c172afd.mp4

> I have no answer. Video is inaccessible.

https://user-images.githubusercontent.com/78635592/169617391-8c20f54e-2b81-4edd-b275-745b1582087b.mp4

dio1985, 2022-05-20:
> > I have no answer. Video is inaccessible.
> 
>  169562382-9c6f0aff-e6b2-4678-90c9-187d3903f28b_1.mp4 
> i just used only english...... file . utility.ino
> 
> String getWeekdayString(uint8_t wd) { String str = ""; switch (wd) { case 1: str = F("Monday"); break; case 2: str = F("Tuesday"); break; case 3: str = F("Wednesday"); break; case 4: str = F("Thursday"); break; case 5: str = F("Friday"); break; case 6: str = F("Saturday"); break; case 7: str = F("Sunday"); break; } return str; }
> 
> I just changed the day of the week to english but it doesn't show properly
and i used "MM dd yyyy dddd" but all dddd or DDDD has something wrong
> 
> "Fri2a22" ??????????

vvip-68, 2022-05-21:
I still have no idea.
Conversion from 'DDDD' or 'dddd' to day of week is in RunningText.ino, lines 1134-1136
Check if your code files are saved in UTF-8 format.

dio1985, 2022-05-21:
> I still have no idea. Conversion from 'DDDD' or 'dddd' to day of week is in RunningText.ino, lines 1134-1136 Check if your code files are saved in UTF-8 format.

I just changed the days of the week in Russian to English. You can find out if you change only one day of the week into English

file . utility.ino

String getWeekdayString(uint8_t wd) {
String str = "";
switch (wd) {
case 1: str = F("Monday"); break;
case 2: str = F("Tuesday"); break;
case 3: str = F("Wednesday"); break;
case 4: str = F("Thursday"); break;
case 5: str = F("Friday"); break;
case 6: str = F("Saturday"); break;
case 7: str = F("Sunday"); break;
}
return str;
}

please try to use english

vvip-68, 2022-05-21:
Yes, really. I got an issue. Because running string processed in cycle until all data-related substitutions are not processed.
So, data-formatter string 'DDDD' initially substituted with 'Fri*d*a*y*', next iteration *'d'* from Friday substituted with day '21' and *'y'* with short-year -- 22...

Will think about way to fix it...

vvip-68, 2022-05-21:
I updated source file runningText.ino
Try to get it and use in your sketch.
I thing issue is fixed.

closed 2022-05-21 by vvip-68

dio1985, 2022-05-21:
> I updated source file runningText.ino Try to get it and use in your sketch. I thing issue is fixed.

wow thank you so much !!i will try to do . 
your project is so great