vvip-68/GyverPanelWiFi #234: auto connection Err

dio1985, 2022-05-22:
https://user-images.githubusercontent.com/78635592/169684103-38c0242b-967b-4d6b-a234-0fda0b23a8b8.mp4

it has something wrong with auto connection. 
first of all , its not working when just only try to connection 
but auto connection tried to connect to phone IP after exit application and reStart application

vvip-68, 2022-05-22:
So what? It senseless test of check connection?
Yes, it is bug. 
When 'Autoconnect' is unchecked, the check of IP address validity performs on button click action. It checked if specified IP differs from IP address received by phone self from router, for preventing establish connection of application to phone self.
When 'Autoconnect' is checked the validation of specified IP address is skipped and connection established as loop to self (like localhost). Obviously in this loop nothing works, because remote device still out of connection and know nothing about data requests from application.

It is so rare and specific situation and I do not want spend my time to modify application for IP address validation if 'Autoconnect' checkbox is checked. You may do it yourself if you wand and have a lot of free time. Source code of application placed in project in folder 'Android/AppInventor'.

vvip-68, 2022-05-22:
During this fascinating process you also may translate labels in application from russian to english. It is very useful case.

closed 2022-05-22 by vvip-68