vvip-68/GyverPanelWiFi #237: Глюк с зависанием игр в игровом режиме при подключенном SD-модуле (#define USE_SD 1)

frol-aleksan, 2022-09-04:
Исправление: в файле a_main.ino в фунцкии
void sendPageParams(uint8_t page, eSources src)
из блока 
case 12:  // Игры
      str = getStateString("UP|FM|W|H|BR|SE|SD");
      break;
убрать SD (оно там по идее вообще не нужно), должно быть str = getStateString("UP|FM|W|H|BR|SE")

vvip-68, 2022-09-05:
А в чем проявляется зависание игр? Никогда не сталкивался.
Этот блок с "case 12:" используется для передачи текущего состояния в приложение WiFiPlayer, в котором используется возможность рисования и сохранения рисунка либо во внутреннюю память, либо на SD-карту, когда она подключена. Если убрать из строки параметров "SD" - чекбокс "куда сохранять" будет недоступен когда карта есть или наоборот виден, когда карты нет (не помню). То есть логика приложения будет нарушена.
В обработке же передачи параметра "SD" приложению - нечему зависать там весь код - добавить к передаваемой строке значение флага isSDCardReady и ничем не отличается от обработки остальных флагов состояния... (a_main.ino строка в районе 3700)

frol-aleksan, 2022-09-05:
Проигрывается несколько кадров и игра останавливается и наблюдается это только при игре, а в деморежиме все нормально. После того, как убрал SD, игры заработали нормально.
Мешать загрузке картинок с карты эта правка в принципе не должна, т. к. за картинки отвечает case 10, а за рисовалку - case 11 со своими строками параметров.

vvip-68, 2022-09-05:
Ну, наверное - да. Было б неплохо, если б вы исследовали истинную причину почему происходит зависание игры. Мне в голову не приходит ни одной причины, почему бы могло такое случаться...

closed 2022-09-05 by vvip-68