vvip-68/GyverPanelWiFi #242: Что-то не пойму логику работы с wdt на esp8266

Spider84, 2022-11-12:
А собственно как?
В коде вижу следующее:
https://github.com/vvip-68/GyverPanelWiFi/blob/163cd6ba6bd7e0465e920aeee4b1b7d7396ba07b/firmware/GyverPanelWiFi_v1.12/GyverPanelWiFi_v1.12.ino#L118

Но более нигде не вижу сброса таймера. Где вызов `ESP.wdtFeed();`?

Или я чего-то не знаю?

vvip-68, 2022-11-12:
А зачем он нужен? Каждый "заход" в loop() сбрасывает таймер (внутри системного кода, который вызывает этот loop). Время выполнения самого loop - заведомо меньше установленного "сторожевого порога" 8 сек - обычно это 20 мс... Простой вызов delay(0) кстати тоже сбрасывает таймер.

Spider84, 2022-11-12:
Где об этом сказано?

vvip-68, 2022-11-12:
Вот источники уже сами гуглите. Читайте документацию. Я за вас это искать не стану. 
Если найдете, что то, что я выше описал - не так. Поделитесь информацией.

closed 2022-11-12 by vvip-68

Spider84, 2022-11-12:
Ну что я могу сказать. Документации на него нет. Сплошной треш. Я пошёл другим путём, к примеру ядро Arduino esp8266 мы же установили в Arduino и оно у нас есть по следующему пути: https://github.com/esp8266/Arduino/tree/master/cores/esp8266
там можно найти файлик [`Esp.cpp` ](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp)в котором обнаружить "[исходный код](https://github.com/esp8266/Arduino/blob/80bf71662551fb5e6579713dc15e541ac48ccf98/cores/esp8266/Esp.cpp#L94)" esp-шного, прости господи, wdt.
Так вот они нигде не вызываются более.
Там же есть файлик [`cores\esp8266\core_esp8266_main.cpp`](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_main.cpp)  в нём есть [`loop_wrapper()`](https://github.com/esp8266/Arduino/blob/80bf71662551fb5e6579713dc15e541ac48ccf98/cores/esp8266/core_esp8266_main.cpp#L239)
который вызывает собственно `loop()` и там нет описанного выше функционала сброса wdt. Даже если пойти выше до `loop_task()` там тоже нет.

Так где? На основании чего сделано заявление?

vvip-68, 2022-11-12:
Исследование исходников - лучший способ понять как оно работает. Согласен.
Почему вы считаете, что та должен быть именно вызов wdtFeed(), Внутри может быть вызов чего угодно. Вплоть до прямой записи в какие-нибудь регистры, которые сбрасывают таймер WDT. Что делается внутри самой wdtFeed()?
Вы реально проверили все цепочку вызовов до loop()?

"Заявление" сделано на основании 
а) просто не помню в каком месте исходников / документации или форумов были подчеркнуты эти сведения
б) по факту того, что оно работает и не перегружается по WDT при штатной работе и, соответственно, перегружается если "застрять" в каком-нибудь цикле на время большее таймера WDT без вызовов  wdtFeed() или delay() что делает то же самое.

Spider84, 2022-11-13:
Да, прошерстил. в AVR реализации нашёл сброс. А в ESP нет. Как-то странно это всё.

Я к чему это поднял. у меня виснет вся эта конструкция спустя 2-3 часа работы колом. Замирает картинка последняя и никакой реакции. Помогает только снятие питания. Так вот судя по тому, что WDT где-то сбрасывается, виснет оно не полностью.

Добавил "костыль", будем посомотреть.
```c
void process() {  

#if defined(ESP8266)
  if (millis()>=3600000U) {
    ESP.restart();
  }
#endif

...
}
```

vvip-68, 2022-11-13:
Ну тут вам скорее всего никто не поможет. Чтобы понять причину и как-то исправить - нужно суметь воспроизвести. У меня и других - ничего не сыплется. И умозрительно по коду тоже вроде нечему сыпаться. Какой размер матрицы, например? Что в это время пишется в логах. Если есть стек ошибки - что говорит о ней Exception Decoder? Что пишется после компиляции проекта - сколько памяти всего/осталось/доступно? Когда "через два-три часа зависает" - что пишется в мониторе порта?

vvip-68, 2022-11-13:
А замирать картинка может элементарно из за плохого контакта / непропая на линии DATA по которой битики бегут к светодиоду. Или в самОм сАмом первом светодиоде, который, к примеру, перегревается и перестает принимать(и передавать дальше) управляющую последовательность. Вариантов - масса. Для начала все же смотрите что в мониторе порта на этом зависании...

Spider84, 2022-11-13:
Там именно зависание. Проц колом. Нет ни даты ни порта.

vvip-68, 2022-11-13:
Перегрев? Нестабильное / некачественное шумящее питание? Помехи / наводки от чего-то рядом стоящего, возможно импульсные? Просто брак конкретного экземпляра микроконтроллера?

Spider84, 2022-11-13:
Модули меня (к слову это ESP-01) есть у меня от одного проекта горсть.  От смены модуля толку ноль. Питание чистейшее как слеза девственницы, смотрел и осцилом, величина шума в районе 10мВ, что менее 0.5%. Хорошие жирные кондёры как керамика 0.1 так и танталы, там уже ёжик был из кондёров когда пробовал.
Стоит она на подоконнике, где шуметь ещё чему-то кроме её собственного БП (железный такой в дырдочку модуль) нечему. Перегрев? Может быть, но когда я прихожу она уже холодная - да и прохладно на окне... Ну можно попробовать вжарить ей радиатор от малинки :)))

vvip-68, 2022-11-13:
Ну, у меня варианты кончились. Я не знаю как еще диагностировать проблему...
Если какие-то фичи в скетче включены (в настройках) - пробуйте отключать по одному и ли отключить все, оставить только вывод на матрицу и смотреть при добавлении какой фичи станет "зависать". Отключите временно смену всех эффектов, выберите один. Если проработает более 2-3 часов, добавляйте другие. Имитации игр включайте в последнюю очередь. Они как были от Гайвера - так я их не трогал, Может в них что. У меня они почти все отключены.

vvip-68, 2022-11-13:
Ну, и многие тут хвалили проект WLED - попробуйте его. Управление - со всех каналов, куча эффектов, работают над ним вроде целая команда, весь код вылизан и вычищен, говорят что стабильная версия. Я не пробовал - это со слов других.