vvip-68/GyverPanelWiFi #245: Ошибка при компиляции

tih66, 2022-11-26:
При компиляции проекта, выскакивает ошибка:
In file included from C:\Users\������������\Documents\Arduino\libraries\FastLED\src/FastLED.h:67,
                 from C:\Users\Пользователь\Desktop\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.13\a_def_hard.h:391,
                 from C:\Users\Пользователь\Desktop\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.13\GyverPanelWiFi_v1.13.ino:77:
C:\Users\������������\Documents\Arduino\libraries\FastLED\src/fastspi.h:145:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  145 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
хотя прошивка компилируется и прошивается. Но при этом матрица не реагирует (не горит не один светодиод в принципе).  Где моя ошибка, не поскажите?

vvip-68, 2022-11-26:
В этом сообщении нет ошибки. Тем более - прошивка компилируется. Не все сообщения оранжевым в логе являются ошибкой.

Матрица не реагирует (не горит) - зависит от множества причин
а) Не соблюдаете рекомендации к прошивке - какие версии ядра и библиотек использовать, какую плату выбирать и т.д.
б) Не к тем пинам подключили матрицу
в) Матрица неисправна
г) Прочее...

Что при всем это выводится в лог в монитор порта?

tih66, 2022-11-26:
Отбой, разобрался. в скетче нужно #define LED_PIN  D2 , а по умолчанию стоит #define LED_PIN  2

vvip-68, 2022-11-26:
Нет, там именно  #define LED_PIN 2 должно быть.
Потому что следующим вопросом буде - "а почему яркость автоматически уменьшается до нуля?"

А потому что - исправили тут, плюс используете или не ту версию ядра, или в Инструменты выбрали не ту версию платы.

tih66, 2022-11-26:
А где посмотреть версию ядра и какое стоит?

vvip-68, 2022-11-26:
Инструменты -> Плата -> Менеджер плат

closed 2022-11-27 by vvip-68