vvip-68/GyverPanelWiFi #247: пропуск кадров Jinx

akrub, 2022-11-28:
Почему то пропускает много кадров при трансляции на панель из Jinx
матрица 15*10
У матрицы включен режим ведомый, на пк запущен Jinx , матрица прописана
Это так и должно быть или что-то можно улучшить?

vvip-68, 2022-11-28:
Нет, такого быть не должно. На всех известных мне реализованных устройствах такого не наблюдалось.
Выявление причин требует глубокого анализа что происходит, который удаленно сделать затруднительно.
Возможно некорректная работа сети. Возможно у вас в сети несколько устройств с одинаковым IP например...
В общем - трудно сказать.

akrub, 2022-11-28:
одинаковых IP точно, нет
IGMP snooping  должен быть включен или нет?

vvip-68, 2022-11-28:
Я не помню. Что мешает поэкспериментировать?

vvip-68, 2022-11-28:
ЛаЮ была ошибка с замираниями. Но оно давно исправлено. Если у вас совсем старая прошивка - то да, такой дефект имел место быть...

akrub, 2022-11-28:
> Я не помню. Что мешает поэкспериментировать?

вечером после работы проверю конечно же, только сейчас просто мысль в голову пришла )

> Да, была ошибка с замираниями. Но оно давно исправлено. Если у вас совсем старая прошивка - то да, такой дефект имел место быть...
по описанию ошибка вроде как была исправления в начале 22 года, а я шился в конце 21 ) тоже проверю сегодня этот момент

отпишусь потом

vvip-68, 2022-11-28:
Хм... А что за режим то?
Насколько я знаю, если включен прием пакетов с джинкса - оно режимы вообще игнорирует и просто принимает всё, что ему шлет мастер... Все остальные настройки вообще игнорируются... Подвернется под руку на чем проверить - посмотрю, но...

akrub, 2022-11-29:
сейчас версия 1.12.1026 , попробую вечером до актуальной обновиться

vvip-68, 2022-11-29:
Если 1026 - это 2022 нод - там принципиально ничего не менялось касательно E131 и синхронизации матриц по сети и с джинксом на компьютере...

akrub, 2022-11-29:
как он может быть 2022 если я собрал и прошил в декабре 2021 года ? )

vvip-68, 2022-11-29:
В декабре активизируются гирляндеры и "Все собрал, ничего не работает" или "Вообще не компилится" - начинают идти потоком. Перечитывать всю историю сверху донизу у кого что - как-то даже времени нет. Просматривается пара последних информативных сообщений, чтобы примерно вспомнить о чем речь.
Версии, когда я их нумерую. пишУтся вроде 1.12.2022.1026 - число и год, когда выпущена. А из того как вы написали 1.12.1026 - это может быть и 1.12.2021.1026, так и 1.12.2022.1026 - об этом и было сообщение.

vvip-68, 2022-11-29:
Кстати, как мне тут написали - в последних версиях (не знаю даже с какой) автоматическое включение авторежима вообще отвалилось... Не включается автоматически. Тоже вот нужно чинить...

akrub, 2022-11-29:
> А из того как вы написали 1.12.1026 - это может быть и 1.12.2021.1026, так и 1.12.2022.1026 - об этом и было сообщение.
да я не в претензии ) извиняюсь пропустил год , почему то показалось что целиком из приложения переписал версию, 2021 там еще перед 1026

vvip-68, 2022-12-04:
Настроил, проверил. На матрице 32x32 = 1024 диода никаких торможений при трансляции с Jinx`а нет.

closed 2022-12-04 by vvip-68