vvip-68/GyverPanelWiFi #248: Добавление своей картинки к эффекту «Анимация»

bdu1977, 2022-11-28:
Решил собрать данный проект еще раз, прошлый был подарен. все собрал, все работает, решил добавить побольше картинок в прошивку, в раздел анимации. прочитал соответствующий раздел Wiki, Использую прошивку 1,13. но в файле animation.ino  не могу найти команды которые указаны в wiki. мои действия были следующие.
-создал картинку moon в программе
-сохранил  код (moon.h)
-скопировал  файл картинки moon.p в папку DATA -->16p16 (на всякий случай)
-скопировал файл картинки moon.h в корень папки со скетчем
-вносил разные корректировки в "картинка.h"
-В файле animation.ino вставил  #include moon.h 
-так же в двух местах  в данном файле добавил по аналогии case

 case 3:
        // Луна 
        loadDescriptor(&animation_moon);
        frames_in_image = sizeof(moon_array) / sizeof(moon_array[0]);
        break;

и  
  case 3:      
      // Луна
      ppFrame = moon_array[frameNum];
      break;

а вот 4 и 5 пункта wiki я не нашел в скетче. попробовал их создать, примерно в том месте где и указано. компиляция прошла без ошибок, а вот в телефоне добавленная анимация не появилась в приложении WiFiPanel  в эффекте "Анимация", в комбобоксе выбора анимации.

Подскажите что делаю не так?

vvip-68, 2022-11-28:
А в каком месте инструкции (и в wiki и в самом комментарии во первЫх строках файла animation.ino) сказано хоть что-то про case? Вы все делаете по какой-то непонятной инструкции, с соответственно с непонятным же результатом...

bdu1977, 2022-11-29:
Ок. все вроде получилось, осталось только посмотреть как выводится на матрицу.
не получалось по другой причине. я открываю скетч, там стоит версия 1,13, но файл animation.ino там совсем другой. скачал сейчас последнюю версию, и все появилось. 
Спасибо огромное.

closed 2022-11-29 by bdu1977