vvip-68/GyverPanelWiFi #249: Некоторые эффекты некорректно отображаются

bomaw, 2022-11-30:
Матрица 14х14 из гирлянды (капля), чип WS2812B, использую ESP-01 (имелись в наличии). 
У меня наблюдаются 2 проблемы:
1. Не горит последний светодиод в матрице, он рабочий 100 % (если указать матрицу 14х15, то загорается).
2. Некоторые эффекты некорректно отображаются, сдвиг на соседних вертикальных линиях, т.е. 1,3,5 и т.д. взять за основу, то 2,4,6 и т.д. сдвиг вниз на 2 светодиода (например часы, бегущая строка, змейка, лабиринт на них очень хорошо это заметно).
Сначала собрал на макетке, сейчас все спаял, пробовал перепрошить.

vvip-68, 2022-11-30:
Что выводится в лог при старте программы?
Указанные симптомы вообще ни на что не похожи. Даже теоретически не знаю что может приводить к таким результатам. Может быть есть у сообщества какие идеи?

bomaw, 2022-11-30:
Дословно не помню, сейчас лог отключен, я использую ESP-01 на ней всего 2 пина (лента, кнопка) + 2 (реле, -) если отключить лог.
Но примерно: 
WiFi подключен. IP ...
UDP-сервер ...
Запуск эффекта: ...
Запуск эффекта: ...
и т.д.
Вечером попробую собрать на макетке с Wemos d1 (выдернуть с др. проекта), может дело в ESP-01.

vvip-68, 2022-11-30:
Врял ли дело в ESP. Налицо явное непопадание адреса светодиода в нужную точку. А значит скорее всего проблема либо с выставленным размером матрицы, либо с типом зигзаг/параллельная.

vvip-68, 2022-11-30:
Схему соединений к ESP-01 красиво нарисовать можете? Чтобы в wiki в список схем вставить.

bomaw, 2022-11-30:
Подключал следующим образом: 
RX(GPOI3) - реле         VCC - 3,3в
GPOI0 - лента              RST - не использовал
GPOI2 - кнопка            CH-PD - не использовал
GND - земля                TX(GPOI1) - не использовал
Для питания использовал понижающий модуль 5V-3.3V с али. Лента через резистор.
Также хотел использовать плату логического согласования уровней, но с ней лента не заработала.
И соответственно исправления в коде подключенные пины (3, 0 и 2).

vvip-68, 2022-11-30:
Может напишите простой скетч - от 0 до NUM_LEDS-1 - по очереди зажигаете светодиод. Задержка. На следующем цикле - следующий диод. Пока не дойдете до конца цепочки. Потом снова с нуля.

bomaw, 2022-11-30:
Сделал по другому, массив из 196 цветов от красного до фиолетового и повторил до конца. Как я и сказал выше, первый теряется. Для проверки в массив спереди добавил белый цвет, отображает все кроме белого. 
Возможно что первый светодиод неисправен или специфика работы данной гирлянды.

bomaw, 2022-11-30:
У меня есть вторая такая же гирлянда (не собрана в матрицу) на ней проверил свой тестовый код, первые 8 горят правильно, дальше не проверял (не стал размазывать). 
Сегодня уже поздно, завтра после работы попробую первый отрезать и припаять один из лишних (было 200).

7918514, 2022-12-01:
> Матрица 14х14 из гирлянды (капля)

Если вы про гирлянду с каплями из "эпоксидки", то это голимое дерьмо. Первый светодиод часто глючит. Такая гирлянда очень требовательна к питанию. Сколько не пробовал их, не получилось заставить их работать нормально. Выбросил и забыл - только время потерял.
Замените ESP-01 на Wemos D1 mini, зачем же себя так мучить?))

bomaw, 2022-12-02:
Вчера поменял первый светодиод, проводил разные тесты. В итоге эта гирлянда с ФИКСИРОВАННЫМИ АДРЕСАМИ светодиодов, видимо первый был не рабочий его "китаец" отрезал, вместо 200 было 199.
Сейчас у меня только 2 варианта (чтобы на НГ была рабочая матрица, новая с Китая не успеет прийти):
1. Пересобрать матрицу, сместить на 1, тогда первый гореть не будет (нет его).
2. Каким то образом исправить код (без глобальной переделки), чтобы было смещение вывода на матрицу.
P.S. На следующий год надо заказать нормальную гирлянду (ленту).

vvip-68, 2022-12-02:
Это не так работает. У диодов нет фиксированных адресов.
На вход цепочки подается длинная последовательность импульсов, кодирующая биты информации - цвета, которыми должен светиться диод. Каждый диод "выкусывает" из этой последовательности 24 первых бита, забирая себе закодированный цвет, остальную цепочку передает дальше следующему в цепочке диоду. И так, пока биты не кончатся.

Попробуйте "поиграться" с фичей, реализованной в версии 1.13 - индексная адресация диодов на матрице. В комплекте в проекте в папке tools есть утилита LedMapper. В ней можно составить "карту" индексов диодов, как они расположены на матрице. Пронумеруйте под ваш случай, подключите пронумерованные диоды (массив карты индексов) к скетчу.
Код массива вставьте в скетч под ваш размер матрицы - в файле index_map.ino.

в Wiki документация еще не доделана, но, надеюсь разберетесь.
https://github.com/vvip-68/GyverPanelWiFi/wiki/Сборная-матрица,-состоящая-из-матриц-разного-типа-и-размера

bomaw, 2022-12-02:
> Это не так работает. У диодов нет фиксированных адресов.

Я тоже так думал. 
У меня гирлянда 199 светодиодов, использую 196, взял 197 припаял вместо первого, запускаю не горит 1 и последний. Написал скетч, загорание последовательно светодиодов от 1 до 200, через каждые пол секунды. Загораться начинает со 2 по 196 после 1.

vvip-68, 2022-12-02:
А у меня тогда вот такой вопрос: А вы точно используете версию ядра 2.7.4, библиотеку FastLED 3.4 или 3.5 и в настройках платы выбрана NodeMCU 1.0?
Версии ядра 3.xxx как раз "славятся" тем, что с ними неадекватно работает первый светодиод в цепочке. Он живет своей жизнью, может моргать произвольными цветам, может вообще не гореть. Правда, чтобы был сдвиг я еще не слышал, ну а вдруг?...

vvip-68, 2022-12-02:
Ну ладно, у вас плата ESP01 и пины в скетче переназначены и работают - значит все ок. Но версия ядра тут принципиальна.

vvip-68, 2022-12-02:
И светодиоды в гирлянде - WS2812B? Или если другие - исправлен ли тип светодиодов в скетче на соответствие реально используемым?

vvip-68, 2022-12-02:
А если в цепочку перед первым диодом припаять "самый первый" (нулевой) диод?

bomaw, 2022-12-02:
Версия ядра 2.7.4, библиотека FastLED 3.5 (из архива).
Пробовал WEMOS то же самое. 
В скетче WS2812B (по факту я уже не уверен, что они именно такие).
Я уверен что дело в гирлянде.
Сегодня попробую взят остатки (2, 199 и 200), спаять, и поиграться с их включением.

bomaw, 2022-12-03:
Эта гирлянда с ФИКСИРОВАННЫМИ АДРЕСАМИ. Подключил оставшиеся 3, размер массив указал 200, по очереди через 1 сек устанавливал цвет элементам массива (1, 198 и 199), они загорались указанным цветом.
С помощь карты адресов удалось заставить работать матрицу правильно (указал начать с 1 по 195 и в конце 0), но последний светодиод так и не горит, также пробовал вместо 0 поставить 196 и добавить к NUM_LEDS +1.
Как заставить работать последний светодиод, в моем случае, пока не разобралься.

vvip-68, 2022-12-03:
Удивительные вещи. Никогда даже не слышал о таких диодах с фиксированными адресами... Чудеса...

bomaw, 2022-12-03:
Адрес его 196, я проверял в тестовом скетче, по этому адресу он работает. 
В Вашем коде поставил карту индексов от 1 до 196, он не работает, поэтому попробовал увеличить NUM_LEDS тоже не работает.

vvip-68, 2022-12-03:
В каком месте меняли? Тут?
[image]

vvip-68, 2022-12-03:
Сейчас в коде проверяется, чтобы индекс диода (начинался с нуля и последний не превышал размеров матрицы W x H - 1. 

Я просто с такими гирляндами никогда не сталкивался вообще и даже не предполагал, что они существуют в природе. Дайте ссылку где покупали?

Ну реально выход получается для диодов с адресами 196,197,198,199,200 нужно в строке 322 где вычисляется NUM_LEDS прибавить нужную длину хвоста адресации.

Или в скетче придется убирать проверку на то, чтобы индекс диодов лежал в 0..NUMLEDS-1 - чтобы был любым...
Если правка строки 322 не поможет - пишите, попробую исправить скетч

vvip-68, 2022-12-03:
А у вас похоже и нумерация диодов в гирлянде не с 0, а с 1 начинается, если всё как вы описали - потому и сдвиг...

bomaw, 2022-12-03:
Менял в a_def_soft.h строка 872.
В файле GaverPanelWIFI_v1.13.ino в строке 322: if (CURRENT_LIMIT > 0) {

bomaw, 2022-12-03:
Гирлянда со своим контроллером, брал [здесь](https://sl.aliexpress.ru/p?key=pmVuxV5), сейчас глянул, есть один отзыв с описанием что адреса фиксированные.

bomaw, 2022-12-03:
Я писал должно было быть 200 светодиодов, а по факту 199, сама гирлянда спаяна из 2-х частей (99 и 100), я предположил, что 0 был бракованный его отрезали, т.к. с их контроллером разница не заметна.

vvip-68, 2022-12-03:
>Менял в a_def_soft.h строка 872.

Здесь эффекта никакого не будет. Это просто объявление переменной и какая-то начальная инициализация.
Реальное значение, которое потом используется - там, где я указал.

>В файле GaverPanelWIFI_v1.13.ino в строке 322: if (CURRENT_LIMIT > 0) {

Ну, значит, у вес несколько неактуальная версия. Но ведь не трудно несколькими строчками выше найти нужно еместо? Это же не принципиально.

bomaw, 2022-12-03:
Точно, не глянул, я думал в одном месте должно быть. Сейчас проверю изменив и это значение.

closed 2022-12-03 by vvip-68