vvip-68/GyverPanelWiFi #250: Некорректное использование не void функции

pilnikov, 2022-12-01:
При компиляции ругается на не корректное использование uint16_t функции getPixelNumber - типа функция же не void, но может и не вернуть ничего в процессе.
сделал так (utility.ino с 289 строчки)

int16_t getPixelNumber(int8_t x, int8_t y) {
  int16_t index = -1;

  if (x < 0 || x >= pWIDTH || y < 0 || y >= pHEIGHT) return -1;

  uint8_t xx, yy, ww, sx, sy, mx, my, mw, snum, num;

  if (sMATRIX_TYPE == 0 || sMATRIX_TYPE == 1) {

    // --------------------------------------------------
    // Матрица состоит из одного сегмента, адресация по углу подключения, типу, направлению
    // --------------------------------------------------

    if (mWIDTH == 1 && mHEIGHT == 1) {
      xx = THIS_X(x, y);
      yy = THIS_Y(x, y);
      ww = THIS_W();

      index = (yy % 2 == 0 || sMATRIX_TYPE == 1)
              ? yy * ww + xx // если чётная строка
              : yy * ww + ww - xx - 1; // если нечётная строка
    }

    // --------------------------------------------------
    // Матрица - сборная, состоит из нескольких сегментов, адресация по углу подключения, типу, направлению
    // --------------------------------------------------

    sx = x / sWIDTH;    // номер сегмента по ширине
    sy = y / sHEIGHT;   // номер сегмента по высоте
    x = x % sWIDTH;     // позиция x относительно сегмента
    y = y % sHEIGHT;    // позиция y относительно сегмента

    // Номер сегмента в последовательности сборной матрицы
    mx = THIS_SX(sx, sy);
    my = THIS_SY(sx, sy);
    mw = THIS_SW();

    snum = (my % 2 == 0 || mTYPE == 1)
           ? my * mw + mx           // если чётная строка мета-матрицы
           : my * mw + mw - mx - 1; // если нечётная строка мета-матрицы

    // Номер дода в сегменте
    xx = THIS_X(x, y);
    yy = THIS_Y(x, y);
    ww = THIS_W();

    num = (yy % 2 == 0 || sMATRIX_TYPE == 1)
          ? yy * ww + xx           // если чётная строка
          : yy * ww + ww - xx - 1; // если нечётная строка

    index = snum * sWIDTH * sHEIGHT + num;
  } else

    if (sMATRIX_TYPE == 2) {
      index = -1;
      int16_t arrIndex = x + (pHEIGHT - y - 1) * pWIDTH;
      // --------------------------------------------------
      // Адресация диодов с сипользованием карты индексов
      // --------------------------------------------------
#if (MATRIX_INDEX == 1)
      // Карта индексов загружена из файла в массив int16_t* imap[]
      if (imap != NULL && x >= 0 && x < pWIDTH && y >= 0 && y < pHEIGHT && arrIndex < NUM_LEDS) {
        index = imap[arrIndex];
      }
#else
      // Карта индексов размещена в скетче в файле index_map.ino как массив const int16_t PROGMEM imap[]
      if (x >= 0 && x < pWIDTH && y >= 0 && y < pHEIGHT && arrIndex < NUM_LEDS) {
        index = pgm_read_word(&(imap[arrIndex]));
      }
#endif
    }
  return index;
}

vvip-68, 2022-12-01:
Ничё не понял, тем более тут номеров строк нет.
Для исправления предупреждения нужно в 360 строку (перед самым выходом из процедуры) поставить
```
return -1;
```

closed 2022-12-01 by vvip-68

pilnikov, 2022-12-01:
да return должен быть. Я просто немножко поменял код, чтоб не ругался. Номера строк относятся к оригинальному utility.ino

pilnikov, 2022-12-01:
int16_t index = -1; перенес в голову функции. везде, где был return - заменил на его на index. И вернул его (index) в конце перед выходом.