vvip-68/GyverPanelWiFi #252: Ошибка с SD картой на ESP8266NodMCU

Maks56rus, 2022-12-03:
Не могу решить проблему с подключением Sd карты. Контролер Esp8266 Node MCU. Модуль sd исполюзую Micro SD Shiald. Когда флешка отсутсвует Все работает, В монеторе порта это все прописывается :
Инициализация файловой системы...
FS: Использовано 16384 из 2072576 байт
Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1103
Свободно ячеек 2992
Host: 'WiFiPanel' >> 20x13
Инициализация SD-карты...
Эффекты Jinx! на SD-карте не обнаружены
Подключение к ASUS_11 -> 192.168.1.118
...............
WiFi подключен. IP адрес: 192.168.1.118
Создание точки доступа PanelAP
Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
IP адрес: 192.168.4.1

А когда вставляю флешку в модуль Sd , Esp8266 зависает и пишет вот такое сообщение: ets Jan 8 2013,rst cause:2, boot mode:(7,0)

waiting for host
Как исправить эту ошибку?

vvip-68, 2022-12-03:
Ну, я бы начал с загрузки примеров, которые идут с библиотекой работы с этим шилдом и добился бы, чтобы тестовый скетч из примера работал стабильно и без сбоев. Если тестовый скетч будет работать - дальше уже смотреть в чем проблема со скетчем гирлянды... Многие SD-shield очень требовательны к питанию карточки.

7918514, 2022-12-03:
> Не могу решить проблему с подключением Sd карты.

Были у меня два бутерброда wemos+sd шилд. Прошивка на них была одинаковая. Однако на одном бутерброде в упор не работала флешка. Перепайка sd шилда ничего не дала. После замены wemosa всё заработало. Такое в моей практике было всего один раз, чтоб wemos косячил. Проверять этот wemos по работе пинов не стал, не было времени возиться с ним.

vvip-68, 2022-12-04:
В интернете пишут два варианта - либо питание "не айс", либо "сопля пайки" в районе пина D8...

Maks56rus, 2022-12-04:
есть еще у меня такой же контролер 8266 Node MCU , к нему попробывал припаяться, все тоже самое происходит, грешу на плату SD Shiald , либо скетче

vvip-68, 2022-12-04:
Еще раз - попробуйте перенести SD_CS_PIN с D8 на другой свободный пин

Maks56rus, 2022-12-05:
Пробывал, результат тот же.  Еще обнаружил что 5 вольтовый контакт на нем не где не прозванивается, где то дефект скорей всего есть в плате. Попробую заменить sd shiald, результат сообщу..

vvip-68, 2022-12-05:
Есть картридеры, которые работают от 3.3 вольта. На вашем есть такой контакт? Туда подается напряжение, если он есть?

7918514, 2022-12-05:
> Есть картридеры, которые работают от 3.3 вольта.

Я так понимаю для SD карты нужно только 3.3В
Если на SD Shield'e нет своего преобразователя 5В->3.3В, то питание 3.3В сразу идёт с контроллера.

Топикстартеру фоточку бы приложить, а то может он 5В пытается на SD карту подать...

Maks56rus, 2022-12-06:
Да вот по этой схеме и подключаю, наприжение как с контроллера брал 3.3 вольта так и отдельно.

Maks56rus, 2022-12-06:
> Как-то сталкивался с таким. Пока на 5V не подал +5В - не работало. Попробуйте (и) его подключить тоже.

одновременно с 3.3В? или отдельно  5В подать?если отдельно просто 5В подать туда не чего не приходит при прозвонке , как будто контакт не где не задействован

Maks56rus, 2022-12-07:
Инициализация файловой системы... 
FS: Использовано 16384 из 2072576 байт
Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1214
Свободно ячеек 2881

Host: 'WiFiPanel' >> 20x13

Инициализация SD-карты...
Загрузка списка файлов с эффектами...
Папка с эффектами /20x13 обнаружена.
Доступных файлов эффектов не найдено
Эффекты Jinx! на SD-карте не обнаружены

вот, теперь вроде читает sd карту, и вроде папку видет на фешке вот только файлы с эфектами не видит..

vvip-68, 2022-12-07:
Скриншот содержимого sd-карточки в этой папке

vvip-68, 2022-12-07:
И, судя по фрагменту лога - это какая-то довольно старая версия...
С какой-то из старых версий была разница в том, как реально файлы созданы. Например файлы '*.out' видела, а файлы '*.OUT' - нет...

Maks56rus, 2022-12-07:
я так понял переписать попробывать файлы с последней версии программы jinx и сохранить в формат out ? ?

vvip-68, 2022-12-07:
А раньше они в чем были? И зачем переписывать?
в sdcard.ino в функции loadDirectory() в строках 63-66 есть проверка
```
fn = file_name;
fn.toLowerCase();
if (!fn.endsWith(".out") || file_size == 0) {
        entry.close();
        continue;    
}
```

Поставь вот до этого всего кода условия вывод в лог в монитор порта что за имя файла считывается

Maks56rus, 2022-12-07:
блин, туплю, не могу найти, скиньте скриншот где поставить  этот код..??

vvip-68, 2022-12-07:
[image]

sdcard.ino

Maks56rus, 2022-12-07:
WiFiPanel v.1.13.2022.1203
Версия EEPROM: 0x5D
Host: 'WiFiPanel'

Инициализация файловой системы... 
Использовано 16384 из 2072576 байт

Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1214
Свободно ячеек 2881

Матрица: плоская 20x13
Адресация: по подключению
  Угол: левый нижний
  Направление: вверх
  Тип: зигзаг

Доступные возможности:
+ Бегущая строка: шрифт 5x8
+ Кнопка управления: тактовая
+ Синхронизация времени с сервером NTP
- Управление питанием
+ Получение информации о погоде
- Управление по каналу MQTT
- Групповая синхронизация по протоколу E1.31
- Дополнительный индикатор TM1637
+ Эффекты Jinx! с SD-карты
- Поддержка MP3 Player

Поиск файлов карт индексов...
Нет файлов карт индексов

Инициализация SD-карты...
Загрузка списка файлов с эффектами...
Папка с эффектами /20x13 обнаружена.
Найдены файлы эффектов:
   2020		177.98 К
   Balls		103.63 К
   Fifteens		121.66 К
   Fireworks		151.70 К
   Jinx!		51.82 К
   Mill		163.71 К
   PostIt!		93.87 К
   Psycho		144.94 К
   Stars		87.86 К
   Wheel		134.42 К
Найденные анимации:
   Сердце	8x8

Поиск слайдов на SD в папке '20p13'...
Поиск слайдов на FS в папке '20p13'...
Поиск слайдов на SD в папке '16p16'...
Поиск слайдов на FS в папке '16p16'...
Слайды не найдены. Будут использованы картинки эффекта 'Погода'

А ВОТ ТАК ПРАВЕЛЬНО ЧИТАЕТ?

vvip-68, 2022-12-07:
Ну, правильно вроде. Эффекты то появились?

Maks56rus, 2022-12-07:
WiFiPanel v.1.13.2022.1204
Версия EEPROM: 0x5D
Host: 'WiFiPanel'

Инициализация файловой системы... 
Использовано 16384 из 2072576 байт

Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1214
Свободно ячеек 2881

Матрица: плоская 20x13
Адресация: по подключению
  Угол: левый нижний
  Направление: вверх
  Тип: зигзаг

Доступные возможности:
+ Бегущая строка: шрифт 5x8
+ Кнопка управления: тактовая
+ Синхронизация времени с сервером NTP
- Управление питанием
+ Получение информации о погоде
- Управление по каналу MQTT
- Групповая синхронизация по протоколу E1.31
- Дополнительный индикатор TM1637
+ Эффекты Jinx! с SD-карты
- Поддержка MP3 Player

Поиск файлов карт индексов...
Нет файлов карт индексов

Инициализация SD-карты...
Загрузка списка файлов с эффектами...
Папка с эффектами /20x13 обнаружена.
Найдены файлы эффектов:
   001		857.27 К
   002		676.51 К
   003		765.75 К
   004		764.22 К
   005		788.63 К
   006		664.31 К
   007		763.46 К
   008		764.98 К
   009		759.64 К
   010		765.75 К
   011		759.64 К
   012		764.22 К
Найденные анимации:
   Сердце	8x8

Поиск слайдов на SD в папке '20p13'...
Поиск слайдов на FS в папке '20p13'...
Поиск слайдов на SD в папке '16p16'...
Поиск слайдов на FS в папке '16p16'...
Слайды не найдены. Будут использованы картинки эффекта 'Погода'
Загрузка файла эффекта: '/20x13/003.out' -> ошибка

БЛ,,,Я А теперь что не так?

vvip-68, 2022-12-07:
А ХЗ.
Вот код:
```
    fxdata = SD.open(fileName);
    if (fxdata) {
      DEBUGLN(F("' -> ok"));
    } else {
      DEBUGLN(F("' -> ошибка"));
      error = true;
    }
```

А почему open() возвращает ошибку - кто ж знает.

Вот тут (чуть выше) 
```
    // При загрузке имен файлов с SD-карты в nameFiles только имя файла внутри выбранной папки -- чтобы получить полное имя файла для загрузки  нужно к имени файла добавить имя папки
    fileName = "/" + String(pWIDTH) + "x" + String(pHEIGHT) + "/" + nameFiles[file_idx] + ".out";
```
Попробуйте в имени файла первый слэш убрать.
Но у других же как-то работает...

vvip-68, 2022-12-07:
Поставьте отладочный вывод после этой строки 
```
DEBUGLN("fileName='" + fileName + "'" );
```

Maks56rus, 2022-12-07:
Инициализация SD-карты...
Загрузка списка файлов с эффектами...
Папка с эффектами /20x13 обнаружена.
Найдены файлы эффектов:
   Psycho		144.94 К
   Stars		87.86 К
   Wheel		134.42 К
   2020		177.98 К
   Balls		103.63 К
   Fifteens		121.66 К
   Fireworks		151.70 К
   Jinx!		51.82 К
   Mill		163.71 К
   PostIt!		93.87 К
Найденные анимации:
   Сердце	8x8

Поиск слайдов на SD в папке '20p13'...
Поиск слайдов на FS в папке '20p13'...
Поиск слайдов на SD в папке '16p16'...
Поиск слайдов на FS в папке '16p16'...
Слайды не найдены. Будут использованы картинки эффекта 'Погода'

Подключение к ASUS_11 -> 192.168.1.118
...............
WiFi подключен. IP адрес: 192.168.1.118

Создание точки доступа PanelAP
Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
IP адрес: 192.168.4.1

UDP-сервер на порту 2390
Таймер включения авторежима: 1 минут(а)
Режим: SD-Карта
NTP-сервер ru.pool.ntp.org -> 213.234.203.30
Отправка NTP пакета на сервер ru.pool.ntp.org

Запрос текущей погоды
Погода получена: Yandex
Город: Totskoye Vtoroye
Сейчас: ясно, -20ºC
Код иконки: 'skc-n'
Цвет неба: '#395ea2'
Сейчас: Темное время суток
Рассвет: 09:24
Закат: 17:15

Загрузка файла эффекта: 'fileName='/20x13/Stars.out'
' -> ok
UDP << ip='213.234.203.30:123'; ntp sync
Получен UDP пакeт размером 48
Разбор пакета NTP
Секунд с 1970: 1670442203
Получено время: 07.12.2022 19:43:23
--------------------
'/20x13/Stars.out' - завершено

ВОТ так должно быть читаться эфекты?

vvip-68, 2022-12-07:
Ну, по логам - работает. На матрице то что-то есть?
И в чем была причина, что в первый раз не работало?

Maks56rus, 2022-12-07:
дефект модуля sd shild , заменил все пошло. К матрице сейчас подключусь и отпишусь есть ли результат.

vvip-68, 2022-12-07:
Автоматическое убавление яркости - не соблюдение требований к версии ядра микроконтроллера - стоит что-то другое вместо требуемого 2.7.4 + (поэтому) переназначение пина вывода на ленту на пин кнопки, который убавляет яркость. Скорее всего будет еще серия глюков типа "эффекты постоянно тормозят и замирают, горид зеленым первый светодиод) и масса других.

closed 2022-12-07 by vvip-68

Maks56rus, 2022-12-07:
я так и понял что из за пина . на D4 изменился, Поменял на D2 значение все в норме. Еще раз спасибо за помощь.

vvip-68, 2022-12-07:
Именно. Но если версия ядр 3.x.x - глюки будут и без смены пина - другие.