vvip-68/GyverPanelWiFi #253: До нового года осталось.....

tih66, 2022-12-09:
На гирлянде пишет что до нового года осталось 22 дня, а везде что 23 дня.
[image] 
строка показа "До {C#00D0FF}Нового года {C#FFFFFF}осталось {C#10FF00}{R01.01.***+}{S01.12.****#31.12.**** 23:59:59}{E21}";

vvip-68, 2022-12-09:
А это вопрос как считать. На самом деле осталось 22 дня 12 часов 52 минуты
Часы и минуты просто не показываются.
Хотите - исправьте скетч чтоб показывались.
Или, если хотите - добавляйте в скетче +1 день к результату.

closed 2022-12-09 by vvip-68

247760, 2022-12-24:
Приветствую @vvip-68, где то на просторах Гайвер группы, кто то писал что по [тут](https://github.com/dbprof/ny-led-matrix/blob/5c018c703792b22f99cf80462bd681ff2b285e50/custom.ino), написан режим COUNTER для отсчёта вроде как от 10 секунд до 0 до Новому году, правда ли это? Спасибо :)

vvip-68, 2022-12-24:
Я не знаком с тем проектом, не знаю что там есть.