vvip-68/GyverPanelWiFi #254: Зависание ардуино при плохой связи с интернетом.

CrazyGhostRider, 2022-12-11:
Здравствуйте
Прошивка v.1.13.2022.1204
Платы Ардуино: ESP 32

Собрал себе вторую гирлянду и повесил на балкон. Но там связь с Wi-Fi плохая (-68 dBm).  Через репитер Ардуино подключаться совсем не захотела. 
Но выявил проблему: Если Ардуино подключилось к интернету, но потом потом интернет пропал (с Wi-Fi соединено), то ардуино  зависает. Точнее — отображает эффекты со скоростью 1-5 FPS.
Такое же зависание произошло и на первой гирлянде (с идеальным сигналом Wi-Fi), когда пропал интернет.

П.с. Пожалуйста, вынесите настройку локального порта 2390 в файл a_def_pass. (При статическом ip можно открыть порт в роутере. И тогда управлять гирляндой со смартфона через мобильный интернет)

vvip-68, 2022-12-11:
Ничего не понял. LocalPort (который 2390) - задается в переменной, которая настраивается из приложения WiFiSetup
Если вам (конкретно вам) требуется задавать ее как-то иначе - ну, меняйте код как вам нужно, я не против.

Для устранения зависаний при пропадании интернета - отключите в приложении все features которые требуют подключения к интернету - синхронизация / получение времени из интернета, получении погоды из интернета, рабту с MQTT. Зависаний не будет.

vvip-68, 2022-12-11:
Наврал, да. со стороны скетча оно фиксированное и не настраивается внешними приложениями. Забыл. Но все равно не вижу связи с выносом куда-то в отдельные файлы типа a_def_pass. В этом файле хранится чувствительная информация - типа логинов / паролей / ключей. При чем тут (то есть там) порт?

vvip-68, 2022-12-11:
Ну, ок. Можно и вынести. В следующей версии.
Вопрос только - а зачем?
Зачем менять настройку порта по умолчанию?

CrazyGhostRider, 2022-12-11:
У меня две гирлянды. И для каждой я задал свой ip и порт. И эти порты открыл в роутере.
Теперь я могу выйти на улицу и через мобильный интернет напрямую подключиться к нужной мне гирлянде. (Не используя MQTT сервер)

vvip-68, 2022-12-11:
Логично, с учетом статического адреса и проброски портов...

closed 2022-12-11 by vvip-68