vvip-68/GyverPanelWiFi #255: Изменение цветовой последовательности для одного отрезка

Melkiyby, 2022-12-11:
Привет. Собрал матрицу 48х14(6х14 на окно) и так совпало что одна лента оказалась отличной по цветовой последовательности. Есть ли возможность для этого участка указать нужную последовательность или нужно глубокое погружение в библиотеку FastLed?
[image]

vvip-68, 2022-12-12:
Мне не известен способ как это сделать и я не вижу простого решения.
Есть вроде какая-то возможность разделять вывод на ленту через подключение к разным пинам контроллера и для каждого пина указать свои настройки. Но это действительно нужно глубокое погружение в FastLED и я с этим не сталкивался и ничего конкретного по этому поводу сказать не могу.

vvip-68, 2022-12-12:
Ну, мои эксперименты показали, что это возможно.
Вывод на части ленты разделяется на разные пины, каждый пин инициализируется со своей цветовой последовательностью. Должно работать.

Melkiyby, 2022-12-12:
Ну про вывод через разные пины это понятно, сейчас ковыряю fastled на возможность внесения коррекции в сам массив что бы еще один провод не тянуть. По результатам отпишусь

vvip-68, 2022-12-12:
Проще всего, видимо, найти процедуру FastLedShow() и в указанное место

[image]

вставить небольшой цикл, который пробегает по участку матрицы, которая "не того цвета" и меняет для каждого пикселя цвета с RGB на GRB или как там оно у вас...

Melkiyby, 2022-12-13:
Всё верно, в это место нужно вставить 
[image]
Осталось проверить на подтормаживание

closed 2022-12-14 by vvip-68

Melkiyby, 2022-12-14:
Был выявлен косяк с морганием изменённого участка(синий с красным чередовались) в режиме рисования, игр, загрузки картинок и на некоторых эффектах. Вот фикс
[image]

vvip-68, 2022-12-14:
Расскажите словами в чем смысл? Что-то не соображу...

Melkiyby, 2022-12-14:
Я согласен что мой метод не самый быстрый. Завтра попробую с memcpy() если конечно мозга хватит.

vvip-68, 2022-12-15:
В проекте есть несколько мест, где используется memcpy(), посмотрите там. Почитайте в инете про функцию с примерами.
Ничего сложного - указывается куда копировать (приемник), откуда копировать (источник) и сколько копировать (в байтах). Помним, что цвета в массиве - эта структура RGB, то есть один цвет - 3 байта. Вроде бы так.