vvip-68/GyverPanelWiFi #256: Циклическая перезагрузка ESP8266 при попытке редактирования эффекта "Вышиванка"

gh0st-0013, 2022-12-12:
Возникает циклическая перезагрузка ESP8266 при попытке редактирования эффекта "Вышиванка" как из mqtt так и при подключении по локальной сети. Пока просто вырезал, что бы не повторялось.

vvip-68, 2022-12-13:
Самое правильное решение.
У меня такой ошибки не наблюдается. 
Исправить то, что я не погу воспроизвести - не представляется возможным.
Другие пользователи вроде тоже не жаловались на это.
Проблема уникальна и наблюдается только у вас.
Попытайтесь поставить утилиту ESPExceptionDecoder и скормить ей то, что выводится в мониторе порта при появлении этой ошибки. Возможно станет что-то понятнее о причинах ее возникновения.

gh0st-0013, 2022-12-13:
Вечером попробую поймать Exception на другой ESP и скормить декодеру...

7918514, 2022-12-13:
У меня вышиванка давно работает без проблем (давным давно была проблема отображения, которую пофиксили).
И на последней прошивке проблем не наблюдается.

vvip-68, 2022-12-13:
Это было пофиксено с год назад - 
adopted/updated by kostyamat

gh0st-0013, 2022-12-13:
> Самое правильное решение.
> У меня такой ошибки не наблюдается.
> Исправить то, что я не погу воспроизвести - не представляется возможным.
> Другие пользователи вроде тоже не жаловались на это.
> Проблема [...]

Скорее всего дело именно в конкретном экземпляре esp. В наличии есть несколько esp, но они отличаются как минимум чипом USB от той, на которой проявляется цикличный ребут. На них же в свою очередь - всё нормально работает. 
Итог: косяк конкретного экземпляра esp.