vvip-68/GyverPanelWiFi #257: Esp32 подключается к точке доступа только один раз

Melkiyby, 2022-12-13:
Может кто то сталкивался с такой проблемой? Если перепрошить с изменением флага eeprom то первый запуск подключается к заданной точке доступа, последующие разы сразу создаёт свою.

vvip-68, 2022-12-13:
Совершенно непонятно что у вас происходит.
Ну да, в скетче так запрограммировано - при прошивке со сбросом флага - создавать точку доступа (если не удалось подключиться к локальной сети). И во второй раз, в принципе, то же самое. 
Чем ваш первый раз отличается от второго?

Melkiyby, 2022-12-13:
Проблема что при последующих подачах питания на контроллер он не пытается подключаться к моей точке доступа, а сразу создаёт свою, а если сбросить флаг eeprom то первый раз подключается

vvip-68, 2022-12-13:
А что у вас за точка доступа? А где и как вы настраиваете куда должен подключаться роутер? Что происходит между первым включением после прошивки и вторым? В чем разница?

vvip-68, 2022-12-13:
И обычно все наоборот - при первом включении после прошивки он создает свою точку, потому что не знает куда подключаться, а при последующих, когда указывается при настройке к какой сети подключаться - пытается уже подключиться к  ней. И только если в течение 5 минут не удается это сделать - создает свою точку доступа.

У вас же сценарий вообще какой-то вывернутый "наизнанку" получается. Что интересно вы для этого делаете и как добиваетесь такого?

Melkiyby, 2022-12-13:
Пока что отмена тревоги, сейчас на работе попробовал и всегда подключается как надо, дома ещё раз перепроверю настройки

Melkiyby, 2022-12-13:
> И обычно все наоборот - при первом включении после прошивки он создает свою точку, потому что не знает куда подключаться, а при последующих, когда указывается при настройке к какой сети подключаться [...]

ну я же правильно понимаю что настроить к какой точке доступа подключаться можно только при компиляции? В приложениях такого функционала я не нашёл.

vvip-68, 2022-12-13:
Неправильно. Есть в комплекте проекта android-приложение PanelSetupWiFi.apk 
И это **основной** способ настройки приложения (куда подключаться, конфигурация матрицы, групповая работа).
А все что меняется в скетче - может и не сработать, потому что зависит от некоторых условий, на которые пользователи обычно плюют.

CrazyGhostRider, 2022-12-13:
Извините, что отвлекаю.
Но как я понял, то "Arduino IDE в меню "Инструменты" -> "Erase Flash" выбрать значение "All Flash Contents"" этого пункта нет для ESP32. И можно только изменять значение константы EEPROM_OK.

vvip-68, 2022-12-14:
Ну изменение EEPROM_OK задумывался как основной метод для того, чтобы указать корректные или нет значения сохранены в памяти. "Erase Flash" -> "All Flash Contents"" - просто упрощенный вариант, потому что некоторые пользователи не могут найти эту EEPROM_OK в скетче и у них "...ничего не работает".

Melkiyby, 2022-12-14:
Очень сильно извиняюсь. Не сразу заметил приложение.

e-resh, 2022-12-15:
Извините вопрос не совсем по теме. Сконфигурировал, прошил ESP8266 - NodeMCU. К ленте не подключал(не знаю, может быть в этом проблема) ! Цепляюсь PanelSetupWiFi.apk подключается, настроил что бы подключалось к роутеру. Еще раз подключаюсь уже по IP в сети, подключается. Но у меня кроме иконок с настройками больше ничего не появляется. Иконок с эффектами и прочим нет или они где то спрятаны что я не могу их найти. В чем может быть проблема?

CrazyGhostRider, 2022-12-15:
Для этого другое приложение "WiFi Panel"

e-resh, 2022-12-15:
Спасибо! попробую.

closed 2022-12-16 by vvip-68