vvip-68/GyverPanelWiFi #258: Не весь список анимаций в приложении

altdaps, 2022-12-13:
Приветствую, спасибо за отличный проект, пользуюсь уже третий год.
Сегодня скачал последнюю версию (и прошивки и приложения), кроме необходимых для работы настроек ничего не менял, но столкнулся с проблемой - в списке анимации в приложении, в выпадающей списке можно выбрать только сердце. В чем может быть причина?

Gomerchik, 2022-12-14:
Так же повторил этот проект на окно с размерами 18*12. Спасибо за труды. Хотелось бы добавить анимацию флага из папки pics, но че-то ни как не догоню как лучше и проще это сделать( Столько программ запутался совсем. Анимацию из gif-ки сделать и запускать с карты памяти? (тогда надо ее прикрутить к проекту), или вписать в скетч? Я скорее копировальщик, чем программист - Хорошая инструкция и окно радует соседей)) 
Подскажите - подмогните, чем сможете.

altdaps, 2022-12-14:
> vvip-68, действительно, марио имеет высоту 16 пикселей, а у меня матрица - 14. В ранних версиях прошивки проверки размера матрицы видимо не было, поэтому работало. Спасибо.

vvip-68, 2022-12-15:
Не было. Зачем вам безголовый Марио?
Впрочем, если хотите - отключите проверку тут:
[image]

vvip-68, 2022-12-15:
> Так же повторил этот проект на окно с размерами 18*12. Спасибо за труды. Хотелось бы добавить анимацию флага из папки pics, но че-то ни как не догоню как лучше и проще это сделать( Столько программ [...]
Подскажите - подмогните, чем сможете.

Хорошая (я так считаю) инструкция по анимациям дана в комментариях в начале файла animation.ino
Количество кадров анимации ограничено 20, для флага, я думаю, хватит. В памяти контроллера это тоже поместится.

Для более сложных эффектов из Jinx, которые занимают много памяти (посмотрите размер файла ролика, сохраненного в Jinx) - требуется подключение SD-карты

Gomerchik, 2022-12-15:
Спасибо. Буду изучать пока жду картридер.

> Так же повторил этот проект на окно с размерами 18*12. Спасибо за труды. Хотелось бы добавить анимацию флага из папки pics, но че-то ни как не догоню как лучше и проще это сделать( Столько программ [...]
Хорошая (я так считаю) инструкция по анимациям дана в комментариях в начале файла animation.ino
Количество кадров анимации ограничено 20, для флага, я думаю, хватит. В памяти контроллера это тоже поместится.
Для более сложных эффектов из Jinx, которые занимают много памяти (посмотрите размер файла ролика, сохраненного в Jinx) - требуется подключение SD-карты
—
 
С уважением,
Станислав Миронов

closed 2022-12-15 by vvip-68