vvip-68/GyverPanelWiFi #259: Новый эффект

waw555, 2022-12-15:
Просьба рассмотреть возможность адаптации и добавления нового эффекта. Для матрицы 16*16 работает отлично, если менять размер матрицы, то все портится.

`#include <FastLED.h>

#define LED_PIN     D2
#define NUM_LEDS    256
#define BRIGHTNESS  255
#define LED_TYPE    WS2812
#define COLOR_ORDER GRB

CRGB leds[NUM_LEDS];
  //WAVE MATRIX 16x16
int matrix[16][16]={
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {255,127,031,000,000,000,000,000,000,000,000,000,000,000,031,127},
  {255,255,127,063,000,000,000,000,000,000,000,000,000,063,127,255},
  {127,127,255,255,127,000,000,000,000,000,000,000,127,255,255,127},
  {000,000,063,255,255,063,000,000,000,000,000,063,255,255,063,000},
  {000,000,000,063,255,255,063,000,000,000,063,255,255,063,000,000},
  {000,000,000,000,127,255,255,127,063,127,255,255,127,000,000,000},
  {000,000,000,000,000,063,127,255,255,255,127,063,000,000,000,000},
  {000,000,000,000,000,000,031,127,255,127,031,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000},
  {000,000,000,000,000,000,000,000,000,000,000,000,000,000,000,000}};

   int shiftY=0;
   int shiftX=0;
   double matrixItem;
   int ledIndex;
 
   int R=0;
   int G=0;
   int B=255;
   
   int tempR=0;
   int tempG=0;
   int tempB=255;

   int colourChange =1;
   
   int sweepCountDelay[2]={0,0};
   int sweepSpeed[2]={3,10};
   
void setup()
{
    delay( 3000 );
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );

}

void loop() {
  
sweepMatrixHrz(); 
    
  for (int x=0;x<16;x++)
  {      
    for (int y=0;y<16;y++)
    {        
      for (int threeCol=0;threeCol<6;threeCol++)
         changeColour();
         
      matrixItem = matrix[(y + shiftY) % 16] [(x + shiftX) % 16];
      if (matrixItem != 0){
        ledIndex = 16*x+(x%2)*(2*y-15)+(15-y);
        matrixItem = matrixItem*BRIGHTNESS/255/255;
        leds[ledIndex].r = matrixItem*R;
        leds[ledIndex].g = matrixItem*G;
        leds[ledIndex].b = matrixItem*B;
      } 
    }
  }
  FastLED.show();
}

void sweepMatrixHrz()
{
     if (++sweepCountDelay[0]>sweepSpeed[0]){
      sweepCountDelay[0]=0;
     if (++shiftX>=16)
        shiftX=0;
     }
     memset(leds, 0, NUM_LEDS * 3);
}

void changeColour()
{
  if (tempR<256 - colourChange)
  {
    B -= colourChange;
    tempB = B;
    R += colourChange;
    tempR = R;
    tempG = 0;
  }else if(tempG<256 - colourChange)
  {
    R -= colourChange;
    G += colourChange;
    tempG = G;
  }else if(tempB<256 - colourChange)
  {
    G -= colourChange;
    B += colourChange;
    tempB=B; 
  }else if(tempB>=256 - colourChange)
  {
    tempR=0;
  } 
}
`

vvip-68, 2022-12-15:
У вас же тут матрица, на основе которой строится эффект - "вбита" в массив 16x16
Как вы себе представляете адаптацию под различные размеры матриц?

vvip-68, 2022-12-15:
Откуда взят эффект? Можно ли где-то посмотреть как он вообще выглядит и стоит ли внимания?

waw555, 2022-12-15:
Да конечно, в открытом доступе лежит.
Исходник и видео. Смотрится очень красиво, у меня матрица 16*16, и я его переделал под 16*16, изначально он 8*16. Я его адаптировать под другие размеры не могу, не хватает знаний.
https://sourceforge.net/projects/ledmatrixspectrumanalyser/

Des-75, 2022-12-17:
это же Нян кэт! :)
Надо брать!

EDUARDT68, 2022-12-18:
Раз уж тут решились попросить касательно эффектов то и я попробую.
Эффект "ДНК". 

// ----------- Эффект "ДНК"
// База https://pastebin.com/jwvC1sNF адаптация и доработки kostyamat
// нормальные копирайты:
// https://pastebin.com/jwvC1sNF
//2 DNA spiral with subpixel
//16x16 rgb led matrix demo
//Yaroslaw Turbin 04.09.2020
//https://vk.com/ldirko
//https://www.reddit.com/user/ldirko/
//https://www.reddit.com/r/FastLED/comments/gogs4n/i_made_7x11_matrix_for_my_ntp_clock_project_then/
 
//this is update for DNA procedure https://pastebin.com/Qa8A5NvW
//add subpixel render foк nice smooth look

void wu_pixel(uint32_t x, uint32_t y, CRGB * col) {      //awesome wu_pixel procedure by reddit u/sutaburosu
  // extract the fractional parts and derive their inverses
  uint8_t xx = x & 0xff, yy = y & 0xff, ix = 255 - xx, iy = 255 - yy;
  // calculate the intensities for each affected pixel
  #define WU_WEIGHT(a,b) ((uint8_t) (((a)*(b)+(a)+(b))>>8))
  uint8_t wu[4] = {WU_WEIGHT(ix, iy), WU_WEIGHT(xx, iy),
                   WU_WEIGHT(ix, yy), WU_WEIGHT(xx, yy)};
  // multiply the intensities by the colour, and saturating-add them to the pixels
  for (uint8_t i = 0; i < 4; i++) {
    uint16_t xy = XY((x >> 8) + (i & 1), (y >> 8) + ((i >> 1) & 1));
    if (xy < NUM_LEDS){
      leds[xy].r = qadd8(leds[xy].r, col->r * wu[i] >> 8);
      leds[xy].g = qadd8(leds[xy].g, col->g * wu[i] >> 8);
      leds[xy].b = qadd8(leds[xy].b, col->b * wu[i] >> 8);
    }
  }
}

void DNARoutine()
{
  if (loadingFlag)
  {
    #if defined(USE_RANDOM_SETS_IN_APP) || defined(RANDOM_SETTINGS_IN_CYCLE_MODE)
      if (selectedSettings){
        setModeSettings(1U+random8(100U), 1U+random8(200U));
      }
    #endif //#if defined(USE_RANDOM_SETS_IN_APP) || defined(RANDOM_SETTINGS_IN_CYCLE_MODE)
    loadingFlag = false;
    step = map8(modes[currentMode].Speed, 10U, 60U);
    hue = modes[currentMode].Scale;
    deltaHue = hue > 50U;
    if (deltaHue)
      hue = 101U - hue;
    hue = 255U - map( 51U - hue, 1U, 50U, 0, 255U);
  }
  double freq = 3000;
  float mn =255.0/13.8;
  
  fadeToBlackBy(leds, NUM_LEDS, step);
  uint16_t ms = millis();
  
if (deltaHue)
  for (uint8_t i = 0; i < WIDTH; i++)
  {
    uint32_t x = beatsin16(step, 0, (HEIGHT - 1) * 256, 0, i * freq);
    uint32_t y = i * 256;
    uint32_t x1 = beatsin16(step, 0, (HEIGHT - 1) * 256, 0, i * freq + 32768);

    CRGB col = CHSV(ms / 29 + i * 255 / (WIDTH - 1), 255, qadd8(hue, beatsin8(step, 60, 255U, 0, i * mn)));
    CRGB col1 = CHSV(ms / 29 + i * 255 / (WIDTH - 1) + 128, 255, qadd8(hue, beatsin8(step, 60, 255U, 0, i * mn + 128)));
    wu_pixel (y , x, &col);
    wu_pixel (y , x1, &col1);
  }
else
  for (uint8_t i = 0; i < HEIGHT; i++)
  {
    uint32_t x = beatsin16(step, 0, (WIDTH - 1) * 256, 0, i * freq);
    uint32_t y = i * 256;
    uint32_t x1 = beatsin16(step, 0, (WIDTH - 1) * 256, 0, i * freq + 32768);

    CRGB col = CHSV(ms / 29 + i * 255 / (HEIGHT - 1), 255, qadd8(hue, beatsin8(step, 60, 255U, 0, i * mn)));
    CRGB col1 = CHSV(ms / 29 + i * 255 / (HEIGHT - 1) + 128, 255, qadd8(hue, beatsin8(step, 60, 255U, 0, i * mn + 128)));
    wu_pixel (x , y, &col);
    wu_pixel (x1 , y, &col1);
  }

  blurScreen(16);
}

EDUARDT68, 2022-12-18:
Вот ссылка на файл с эффектами из проекта откуда я взял код

https://drive.google.com/file/d/18fDhEJgJZGMLG5n_ylB0PgWp91Dvn7ag/view?usp=share_link

Код эффекта ДНК под строкой 6205

За ранее благодарю.

kostyamat, 2022-12-18:
Не трогали бы эффекти от Турбина, неадекватный он. Мы в результате все его эффекты удалили, часть переписали с ноля. От греха подальше.

EDUARDT68, 2022-12-19:
[В Этой?](https://github.com/vvip-68/GyverPanelWiFi)

closed 2022-12-22 by vvip-68