vvip-68/GyverPanelWiFi #261: До Нового года осталось - дальше отображаются фигурные скобки и то, что в них (макросы вшиты из примеров)

ACKET2009, 2022-12-24:
Здравствуйте. 
 До недавнего времени макросы, вроде показывали, сколько осталось дней - 11, 10... А теперь первый длинный макрос пишет с изменением цветов строку, но отображает и фигурные скобки и текст за ними. Я ничего не менял - это как-то само изменилось. Может нужно что-то изменить в макросах? Или почему такое происходит?

247760, 2022-12-24:
@ACKET2009 вот бы ещё скрин\копия строки из телефона. Там сначала идут дни, потом новое условие дни и часы, видимо у Вас лишняя случайная скобка и начала светиться в виду выполнения нового условия.

vvip-68, 2022-12-24:
Возможно макрос был не совсем корректно написан изначально. И в той версии такие "отступления от правил" сходили с рук. По "многочисленным просьбам трудящихся" этот макрос немного передела - вместо конкретного года теперь можно указывать `****` или `***+` плюч время задавать с секундами. Возможно отступление от стандарта написания макроса уже не позволяет скетчу его правильно интерпретировать...

ACKET2009, 2022-12-24:
[image]
[image]
Вот такие макросы подтянулись при прошивке из скетча. Я ничего не менял...

ACKET2009, 2022-12-24:
Я не сам писал - это примеры макросов, которые встроились из скетча при прошивке.

vvip-68, 2022-12-24:
У меня точно такие же макросы - все прекрасно отображается.
Второй макрос - "особенный" он при нажатии на кнопку "Показать" сам остаток времени не покажет.
Он полностью работает только внутри скетча, когда приходит время ему отображаться - там вычисляются все нужные параметры, которых при отображении по кнопке "Показать" просто нет в наличии.

ACKET2009, 2022-12-24:
странно... я на две платы загрузил. Одна на окне 12х12 диодов на mini 8266, другая на 16х16 диодов матрица на nodemcu и на обеих одинаковая история...

vvip-68, 2022-12-24:
Ну дело же не в самих платах, а в том, что где-то что-то неправильно настроено или работает. И если для обоих плат (пусть и разных) настроено / работает одинаково неправильно - результат будет одинаково неправильный.

Почему именно у вас происходит вот так - я не знаю. Даже предположений никаких нет. Может кто-то другой сталкивался с этим и решил проблему. Я в этом помочь не могу.

Посмотрите в мониторе порта. Обычно если что-то идет не так, в мониторе выводятся какие-нибудь сообщения в лог.

vvip-68, 2022-12-24:
Попробуйте в этой строке там где в макросе #C - прямо на телефоне стереть "C" и написать еще раз заглавную латинскую "C". Может быть в скетче по ошибке или недоразумению в готовых текстовых строках вместо латинской С закралась точно такая же по написанию, но буква кириллицы "C"?

ACKET2009, 2022-12-24:
Ну это, получается, только с этим макросом, который на первом скриншоте... остальные все хорошо работают... Я пока внешне не могу определить правильность их написания - нужно сверяться со справкой. Но если они написаны правильно, то в каком месте копать? Может в последнем скетче присутствуют поправки не совместимые с шаблонами, заложенными изначально?

ACKET2009, 2022-12-24:
Хорошо, попробую.

vvip-68, 2022-12-24:
У меня этот макрос выглядит на телефоне точно так же как на скриншоте и он работает.

ACKET2009, 2022-12-24:
пробел поменял после До - убрал и после первого макроса поставил. Так заработало...

vvip-68, 2022-12-24:
[image]

closed 2022-12-24 by vvip-68

7918514, 2022-12-25:
Друг столкнулся с подобной проблемой и именно вчера. Причём у него на двух устройствах одинаково загружено (прошивка+макрос). На одном устройстве норм работает, а на втором шпарит со всеми скобками и символами. Сравнили - отличий не нашли. Скопировал с рабочего, вставил в нерабочий, сохранил, всё заработало штатно. Причём до вчерашнего дня всё работало без нареканий. Может быть простое нажатие "Сохранить" решило бы проблему, но увы сначала скопипастили, а потом "Сохранили".

vvip-68, 2022-12-26:
> А какого числа должно показывать помимо остатка дней ещё и часы?

Конда останется 0 дней

ACKET2009, 2022-12-26:
Странно эта строка себя ведёт - опять полезли символы... убрал первый цветовой макрос, в итоге... так стабильно показывает...