vvip-68/GyverPanelWiFi #262: При включении тухнущие красные часы, пока не перевключишь несколько раз эффекты не запускаются

ACKET2009, 2022-12-24:
Здравствуйте! При включении питания начинает работать последний запущенный эфект, потом он притухает и запускаются красные часы у которых то минуты показывает, то двоеточие, то температуру то есть как чеширский кот - то одно появится, то другое... В общем пару раз перевключишь и, вроде нормализуется всё... Это из за чего такое?

vvip-68, 2022-12-24:
Неверно выбранная версия ядра микроконтроллера или неверно выбранная плата в настройках компиляции

closed 2022-12-24 by vvip-68

ACKET2009, 2022-12-24:
Всё, как по инструкции выбрано. Одна из плат Wemos D1 mini, другая NodeMCU, обе установлены, как  "NodeMCU 1.0(ESP-12E Module)" при компиляции, установки в скетче, соответственно выбрал под ESP8266...

vvip-68, 2022-12-25:
Всё по инструкции, говорите?
Ну вот она причина то:

[image]

Читайте внимательно инструкцию!

vvip-68, 2022-12-25:
[image]

7918514, 2022-12-25:
@ACKET2009 **Upload Speed** ставьте по максимуму - это значительно сократит время заливки прошивки в контроллер.