vvip-68/GyverPanelWiFi #265: Приложение при коннекте с ESP32 не показывает полный функционал

VITYQ, 2022-12-27:
## Устройства
- Контроллер - Esp32
- Роутер - Keenetic
- Смартфон - Pixel 4 на Android 13

## Кейс
-> Подключаю esp32 к вайфаю через приложение - успешно
-> Подключаюсь с телефона через домашний вайфай к esp32 - успешно
-> Приложение не отображает полный функционал

## Проблема
Приложение не отображает полный функционал при подключенном esp32 к wifi, а только как при привязке с точки доступа (доступны только 4 пункта меню)

## Логи
Лог при подключении к вайфаю
```
Подключение к Keenetic-6226 -> 192.168.1.48
..........
WiFi подключен. IP адрес: 192.168.1.48

UDP-сервер на порту 2390
Переход в авторежим отключен
Режим: Тени
NTP-сервер ru.pool.ntp.org -> 193.192.36.3
Отправка NTP пакета на сервер ru.pool.ntp.org
UDP << ip='193.192.36.3:123'; ntp sync
Получен UDP пакeт размером 48
Разбор пакета NTP
Секунд с 1970: 1672128263
Получено время: 27.12.2022 08:04:23
--------------------
UDP << ip='192.168.1.42:2390'; cmd='$18 0;'
Получен UDP пакeт размером 6
Ответ на 192.168.1.42:2390 >> ack0;
...
```

Заранее спасибо

vvip-68, 2022-12-27:
Вы не то приложение используете.
Каждое из трех приложений при подключении (если коннект состоялся) отображает свой полный функционал и никаких ограничений дополнительных нету.

closed 2022-12-27 by vvip-68

VITYQ, 2022-12-27:
Я использую приложение PanelSetupWiFi.apk, тогда какое необходимо использовать?

vvip-68, 2022-12-27:
Для настройки матрицы - PanelSetupWiFi. Для управления матрицей - GyverPanelWiFi. Для игр и рисования - GyverPlayerWiFi.

VITYQ, 2022-12-27:
Как-то сложно)
Можно было бы в одно объединить)
Если хотите, можем организовать, я андроид разработчик
Спасибо за ответ

vvip-68, 2022-12-27:
Но ведь если что-тог сделано - это сделано по какой-то причине, неправда ли?
В данном случае все приложения написаны в блочном конструкторе AppInventor (до этого - в Thunkable).
После того как такое приложение-конструктор достигает определенного размера - эти конструкторы просто перестают его собирать с ошибкой "слишком много блоков". Вторая причина - по мере увеличения размеров такого приложения в конструкторе - AppInventor начинает жутко тормозить и с делать в нем какие-то изменения (не говоря уже о разработке нового функционала) - дикая головная боль. Один щелчок или движение мышкой и 30-40 секунд ожидания реакции. А может и больше. Можете сами попробовать - "исходники" в проекте есть.

Писать настоящие андроид-приложения я не умею
Если возьметесь - это будет здорово. Описание API есть в wiki. Если возникнут вопросы - я проконсультирую.
Все на безвозмездной добровольной основе. Возьметесь?

VITYQ, 2022-12-27:
Было бы интересно поучаствовать на безвозмездной основе в развитии Вашего приложения как сайд-проект. Давайте я пока для начала изучу api и в течение дня-двух дам ответ)

VITYQ, 2023-04-01:
@vvip-68 свяжемся по тг? По нику @VITYQ можно найти меня. Я начал разрабатывать мобильное приложение, есть вопросы)