vvip-68/GyverPanelWiFi #266: После включения горят часть диодов, некоторые мигают. И все

ar4ivarius, 2022-12-27:
Добрый день! Установил все необходимые библиотеки (в т.ч. из архива), версии ядер. Откорректировал параметры прошивки под себя: размер, угол подключения, направление, тип т.д. Никаких дополнительных блоков в системе нет (SD, кнопок и т.д.). Прошивка компилируется и заливается без ошибок. Но происходит только то, что указано в теме. при этом все 3 ваших приложения видят гирлянду и подкл. к ней. Куда копать?

vvip-68, 2022-12-27:
Думаю наиболее вероятное направление - в сторону блока питания. Блок должен быть без шумов в линииЮ достаточной мощности и напряжение питания на выходе - не более 5.2 вольта

ar4ivarius, 2022-12-27:
Дело в том, что на прошивке Alex Gyver все работает. Напряжение на блоке 4.98 - меньше регулятор не дает выставить

vvip-68, 2022-12-27:
Это напряжение - оптимальное. По миганиям - тогда ничего сказать не могу. В прошивке ничего принципиально отличающегося от прошивки Alex Gyver нет. Всё то же самое. Значит и работать должно так же.

ar4ivarius, 2022-12-27:
Эх, жаль: сам я не допетрю)) Уточню на всякий: горят одним цветом (красным). Мигаю это значит тухнут небольшая часть из горячих. Пробовал управляющий подкл и к D2 и к D4

vvip-68, 2022-12-27:
Не может оно не работать, если все собрано правильно и версии ядра и библиотек, как вы пишете правильные. Все работа устройства - сформировать в массиве в памяти триады цветов и выплюнуть всё это на ленту. Это и у Гайвера так, и в этом проекте и в сотне других. Принцип один. Проверяете еще раз что собрали или сделали не так. Но раз вы говорите, что "се сделал правильно", версии ядра правильные, подключено к правильным пинам - значит должно работать правильно. Если не работает - значит где-то что-то сделали не так. Тем более никаких других устройств нетЮ только контроллер и лента.

Какой конкретно микроконтроллер используется? Диод синий горит, когда по логам в приложении пишет, что подключилось и работает. Приложите скриншот настроек в развернутом меню IDE Инструменты-Выбор плат

ar4ivarius, 2022-12-27:
Вот недаром говорят: семь раз отмерь, один раз отрежь. Я 2 раза проверял эти настройки. Сейчас решил проверить третий и увидел, что для Erase Flash стояло Only Scetсh...

В любом случае навели вы))

Огромное спасибо!

closed 2022-12-27 by ar4ivarius