vvip-68/GyverPanelWiFi #267: Тормозит!!!

Almido2022, 2022-12-27:
Wemos D1 mini - все прошил. Изначально в  скетче особо ничего не менял, подключил готовую матрицу 16Х16  - все работало: и через точку PanelAP и при подключении к моей сети. 
 Потом в скетче изменил размер матрицы, угол подключения и т.д.  Прошил и... начались проблемы. В общем, ситуация следующая:  если я просто подключаюсь к PanelAP в приложении, то все отлично работает., Но стоит мне зайти в WiFi Panel Setup, указать там сеть, пароль, и IP адрес, плата подключается к моей сети, но начинает капец как тормозить - строка с моим IP адресом "едет" несколько минут (про подключение приложения, эффекты и т.д. я вообще молчу)
Уже и перепрошивал несколько раз, и плату другую пробовал - ничего не помогает. В чем может быть проблема?

vvip-68, 2022-12-28:
Проблема в отсутствии доступа устройства к интернету. Или неправильно настроен адрес (не в той сети, что роутер), или неправильно настроен роутер, что не пускает в интернет. Этот проект получает время и погоду из интернета, и если его нет - отключайте все возможности изделия, которые требуют доступ в интернет.

Almido2022, 2022-12-28:
Спасибо огромное! Буду пробовать.

closed 2022-12-28 by vvip-68