vvip-68/GyverPanelWiFi #268: Сменить MAC

badfly, 2022-12-28:
Здравствуйте! есть ли возможность сменить мак адрес платы? Имею 2 матрицы на окнах на Wemos d1 mini. Роутер их путает, потому как идет 1 мак на обоих платах.  Поменял мак принудительно перед прошивкой, после прошивки панели он опять на дефолтный сменился, который одинаковый.

frol-aleksan, 2022-12-28:
https://circuits4you.com/2017/12/31/how-to-change-esp8266-mac-address/
Попробуйте сделать по этой инструкции. А вообще странно, это получается, что китайцы стали жульничать?

badfly, 2022-12-28:
> https://circuits4you.com/2017/12/31/how-to-change-esp8266-mac-address/ Попробуйте сделать по этой инструкции. А вообще странно, это получается, что китайцы стали жульничать?

Я менял мак перед прошивкой похожим методом, он изменился. При перепрошивке на матрицу мак поменялся на старый.

vvip-68, 2022-12-28:
Так я так понял этим способом адрес меняется программно из скетча. Вы в скетч гирлянды перенесли (добавили) строчки смены адреса из скетча, указанного в  этой статье?

badfly, 2022-12-28:
Не добавлял. А в какое место их добавить можно? я с кодом на вы, ардублоком программирую только.)

vvip-68, 2022-12-28:
В то же, где они в скетче примере. То есть объявление массива MAC-адреса перед setup() сам вызов функции установки мак-адреса - в setup()

closed 2023-01-01 by vvip-68