vvip-68/GyverPanelWiFi #271: Сильные помехи на матрице на новой версии прошивки.

CrazyGhostRider, 2022-12-29:
Здравствуйте
Плата Ардуино: ESP 32
Проблема 1
Решил обновить прошивку с v.1.13.2022.1212 на v.1.13.2022.12214
И появились помехи в отображении картинки. Сам эффект или бегущая строка частично видна, но вокруг много глюков.
Откатился на v.1.13.2022.1204 — всё работает. Потом на  v.1.13.2022.1212  — также работает.

Проблема 2:
https://dropmefiles.com/7Ofzz
В чем ошибка в этих файлах настройки, что после перепрошивки ESP32 помнит все режимы и настройки. Кидал файлы в только что скаченную прошивку. Кроме добавления этих файлов настройки я ничего не менял.
А когда использую только "a_def_hard.h" и "a_def_soft.h" (Перенеся все настройки из "a_def_devc.h" и "a_def_pass.h". И удалив  эти файлы и A_DEF_DEVC=0, A_DEF_PASS=0) То ESP32 обнулена и нужно загрузить сохраненные настройки в приложении.
"EEPROM_OK" всегда меняю.

vvip-68, 2022-12-29:
1. Отличие версии 1214 от 1212 состоит в том, что
а) добавлен КОММЕНТАРИЙ как сделать вывод на два пина для лент со специфической адресацией диодов
б) в эффекте Траффик ярко светилась последняя точка хвоста змейки - добавлено условие проверки яркости
в) Эффект Штора - не запускался потому что не обнулялся счетчик - добавлено обнуление.

Эти изменения НИКАК не могли повлиять на стабильность работы матрицы и наличие глюков. Да и вообще изменения последнего года никак не могли на это повлиять.

"Вокруг много глюков" - скорее всего либо шумящий блок питания, либо превышение напряжения питания уровня 5.2 вольта без наличия в конструкции подпорного диода.

2. Ошибка видимо в том, что в вашей версии файла есть EEPROM_OK, которая грузится компилятором ДО загрузки основного файла где она определена в скетче и эта последняя загрузка переопределяет значение того, что вы изменили в собственном файле.

Сначала грузится a_def_hard.h, из него грузится d_def_devc.h и затем a_def_pass.h и только после этого - a_def_soft.h, в котором последний #define EEFROM_OK  переопределяет ранее заданные значения в других файлах.

CrazyGhostRider, 2022-12-29:
1. Я сам удивился, что не должно было ничего сломаться. Но два раза пробовал v.1.13.2022.12214 и были глюки.
Напряжение выставлено 4.8v
Но из-за чего же только на новой версии прошивки глюки?...

2. Значит всё равно надо в файле a_def_soft.h менять EEFROM_OK.

vvip-68, 2022-12-29:
1. У меня нет ответа. По моему разумению - этого не может быть. И у меня, к слову, такого не наблюдается.
2. Ну, можно просто прошивать с опцией Erase All Content что делает то же самое

CrazyGhostRider, 2022-12-29:
1) Могу попробовать постепенно добавлять/откатывать изменения в прошивке, чтобы выявить глюк. Можно ли как-то посмотреть все изменения?

2) Не вижу этой опции. 
[image]

vvip-68, 2022-12-29:
1. Будет интересно. Ну, например, поставить что-то типа SmartGit, который с каждым коммитом показывает какие файлы были изменены и чир конкретно в файлах менялось от версии к версии. Возможно есть еще какие-то утилиты. Не знаю. Я пользуюсь SmartGit
2. А, чёрт. У ESP32 этой опции нету... Увы...

CrazyGhostRider, 2022-12-29:
Выявил, что изменения от 21 декабря 2022г дают глюки. До этого все было нормально.

vvip-68, 2022-12-29:
А какие именно изменения не удалось выявить?
Сортировка массива файлов при старте программы, считанных с SD карты?
Выравнивание пробелами ширины вывода этого списка после сортировки?
Чтение файла не с реальной SD а с файловой системы контроллера когда включен эффект "SD-карта"?
Исправление опечаток в комментариях? Перестановка (группировка) строк #define чтоб по теме настройки ближе были,
Изменение строки номера версии?

Что из этого может давать помехи? А больше там никаких изменений не было...

CrazyGhostRider, 2022-12-29:
Извините за ошибку.
Версия 14 декабря 2022г работает. А от 20 декабря 2022г не работает.

Вот все настройки
https://dropmefiles.com/Avp84

vvip-68, 2022-12-30:
А мы точно за этот проект говорим? Была версия от 14 декабря, следующая - от 21 декабря. От 20 декабря вообще версии не было.

[image]

vvip-68, 2022-12-30:
[image]

vvip-68, 2022-12-30:
Да. вижу была одна версия, обозначенная как от 20 декабря.
В ней добавлена сортировка массива считанных с SD карты файлов эффектов для эффекта "SD-карта"
Сортировка выполняется в setup(). У вас вообще, судя по приведенному файлу настроек SD-карты в системе нет, а значит этот (измененный) кусок кода вообще компилятором выбрасывается и в сборке проекта не участвует. То есть с точки зрения компилятора для вашей конфигурации различия не работающей версии от 20 декабря  по сравнению с работающей версией от 14 декабря - это изменение строки (константы) с номером версии в главном файле скетча. Всё.

В общем - я не знаю чем вам помочь. Увы. Оставайтесь на той версии, которая работает.

CrazyGhostRider, 2022-12-30:
Понятно
Позже попробую ещё покопаться.

vvip-68, 2022-12-30:
Кстати, достал свою ESP32, залил последнюю версию прошивки - все работает стабильнейше, впрочем, как и ожидалось...
https://drive.google.com/file/d/1r-KN-z5dy2sOn2uAa-_OzXylgSAmQlx5/view?usp=share_link

vvip-68, 2022-12-30:
Вопрос - а версия ядра то у вас какая? А то, может, вы тоже из этих, из любителей поставить  последние версии, чтобы посвежее в ущерб стабильности?

CrazyGhostRider, 2022-12-30:
Arduino 1.8.13

vvip-68, 2022-12-30:
У... Вы, похоже, версию ядра от версии ардуино не различаете, выходит?

CrazyGhostRider, 2022-12-30:
Терминологию — да, плохо.
Но посмотрел, о чем вы.
FastLED 3.5.0
ESP32 1.0.6

п.с. приболел. Мозги не варят.

vvip-68, 2022-12-30:
С версиями все в порядке...
Выздоравливайте...

vvip-68, 2022-12-30:
Не думаю, что это на что-то влияет. Но у меня версия ардуино IDE - 1.8.19...

closed 2023-01-02 by vvip-68