vvip-68/GyverPanelWiFi #272: Проблемы с кнопкой

P0weredUP, 2023-01-29:
Тип кнопки- физическая. В прошивке установлена. Использую вемос. Аппаратно с кнопкой все в порядке, пробовал разные. Подключена между D4 и GND. В прошивке PIN_BTN 2 и LED_PIN D2.
Одиночное нажатие работает нормально, матрица включается и выключается. Удержание-тоже. А вот при двойном и более нажатии начинает произвольно меняться яркость, помогает перезагрузка кнопкой на плате. Следующий эффект кнопкой ни разу не удалось включить. Как фиксить?

vvip-68, 2023-01-30:
Собрать проект с рекомендованной версией ядра 2.7.4? а не с первой попавшейся.

P0weredUP, 2023-01-30:
Ядро теперь стоит 2.7.4 и кнопку и в скетче и физически переместил на D3. Пробоема не решилась

vvip-68, 2023-01-30:
Тогда не знаю. Не сталкивался с таким. У меня нет ответа.

vvip-68, 2023-01-30:
А библиотека кнопки из комплекта проекта стоит или случайная откуда-то установленная?

P0weredUP, 2023-01-30:
Всё как из папки проекта взято , так непосредственно в папку libraries и скопировано. Там даже других библиотек нет, только из этого репозитория

P0weredUP, 2023-02-01:
Пока так и не нашел решения проблемы. Буду искать позже. В любом случае, спасибо за наводки

7918514, 2023-02-04:
> Подключена между D4 и GND. В прошивке PIN_BTN 2 и LED_PIN D2.

Это как это так?
Если использовать рекомендованное ядро 2.7.4, то:
Пин ленты нужно указывать без D
Остальную перефирию с буквой D
В вашем случае так:
#define LED_PIN 2
#define PIN_BTN D4

P0weredUP, 2023-02-04:
Видимо написал это в процессе поиска решения. Сейчас стоит 2 и D3, проблема не ушла. При двойном клике яркость начинает постоянно уменьшаться, когда доходит до конца- перескакиевает в максимум и так циклично до перезагрузки. Вкл/выкл по клику и яркость удержанием работают нормально

7918514, 2023-02-04:
Очень странное поведение. Бывают проблема с кнопкой когда указывают сенсорную, а используют тактовую.
Замените саму кнопку, больше мыслей нет. Ну а вдруг!?...

P0weredUP, 2023-02-04:
BUTTON_TYPE 1 Стоит. Кнопок уже штуки 3 было попробовано, а сейчас оно собрано всё, да и вряд-ли поможет

P0weredUP, 2023-02-04:
Ещё информация: Через сериал монитор узнал, что двойные/тройные/четверные нажатия определяются. Бегущая строка по четверному клику включается, с яркостью всё окей. Тройным кликом переключается эффект (видимо на предыдущий?), тоже не заметил проблем. При первом после включения двойном нажатии оно отображается в порт, но через небольшое время (меньше минуты) начинает медленно циклично угасать. Эффект не меняется. При последующем двойном клике скорость этого угасания увеличивается

vvip-68, 2023-02-05:
Насколько я помню - двойной клик - переключение эффекта на следующий (*он и остается работать), при тройном клике - включение режима автоматической смены эффектов.

Описанного вами поведения в прошивке не запрограммировано.
Могу только развести руками и предложить в a_main.ino закомментировать следующий кусочек кода:

[image]

P0weredUP, 2023-02-05:
Во первых, поправлю себя. Тройной клик таки переключает на случайный эффект.  Теперь к проблеме. Я добрался до строк 430 и дальше и начал пробовать. Сначала думал просто поменять местам клики 2 и 3. Но потом выяснил, что проблема переходит на тройной клик. Как выяснилось- проблема где-то в nextMode();, потому что заменив ее на вызов setRandomMode(); с яркостью проблем нет. Меня выбор случайного режима устраивает, может это даже лучше. Пока у меня и двойной и тройной будут выбирать случайный эффект, разница только в демо будет, а позже полезу поглубже в код, может чего пойму. Спасибо за подсказку, где рыть

P0weredUP, 2023-02-05:
А вот фиг там, автосмена ведь тоже на вызове nextMode работает, получается, что при демо тоже надо случайный выбирать

vvip-68, 2023-02-05:
Ну вообще-то nextNode() просто меняет текущий режим - в зависимости от флага "вкл.случайный выбор" - либо на следующий, либо на случайный, и никакого отношения к изменению яркости не имеет...

Даже не знаю что тут еще сказать...

P0weredUP, 2023-02-05:
У меня было define smooth_change 1
Поставил 0, пока всё в поряде

vvip-68, 2023-02-05:
О, я этим никогда не пользовался. Осталось еще со времен первичного проекта от Гайвера. Даже не знаю работает оно или нет, но по-видимому - не работает... Надо его совсем убрать, что ли...

P0weredUP, 2023-02-05:
Всё, наконец то. Вернул всё, что менял назад и выключил нафиг этот smooth change, переживу и без него, всё работает как задумано. Спасибо за помощь

closed 2023-02-05 by P0weredUP

vvip-68, 2023-02-05:
В следующей версии его вообще не будет...

P0weredUP, 2023-02-05:
Ещё один вопрос, уже не по теме, но открывать ещё одну не хочу. Во время рисования пиксели не успевают бстро рисоваться на матрице. Т.е. я провожу линию за секунду, а на матрице оно появляется медленнее, секунды 2-3. Это ограничение системы или только у меня так?

7918514, 2023-02-05:
У меня так же при рисовании, с запозданием. Бывает даже какие-то пиксели не прорисовываются на матрице, хотя на экране телефона всё норм. Для этого в программе-рисовалке есть даже кнопка - перенести всё на матрицу с экрана.