vvip-68/GyverPanelWiFi #274: Многопоточность для применения нескольких mac адресов одновременно

cochelevalexander, 2023-02-03:
Здравствуйте! В процессе работы стало интересно, есть ли возможность разделения подключения на два и более потоков? Для одновременной работы, к примеру, 14 и более вселенных

vvip-68, 2023-02-04:
Ну на ESP32 по крайней мере 2 ядра. Изучите программирование в FreeRTOS, напишите скетч, поддерживающий многопоточность и будет вам счастье.

closed 2023-02-04 by vvip-68

vvip-68, 2023-02-04:
А просто в коде указать, что MAX_UNIVERCE_COUNT не 7, а 14 - что - не работает?

cochelevalexander, 2023-02-04:
Падает FPS. Вообще, идея в том, чтобы заставить переподключаться один процессор на 2 mac адреса. Т.е. принимать сначала 7 вселенных, оставлять их в памяти, после загружать оставшиеся с 8-14 вселенных, а после сделать одновременный вывод на 2 пина. ( Таким образом, один пин будет отвечать за матрицу 4х4).

vvip-68, 2023-02-04:
А с чего сначала получение одной вселенной, потом другой (последовательно, т.к один процессор (ядро)) ускорит процесс? С чего вдруг?

Ну а параллельный вывод на два пина (да хоть на 4) - кто вам мешает сделать прямо сейчас?
Пишите там где инициализация ленты - делите массив светодиодов на сегменты, каждый сегмент - на свой пин кидаете.

```
  FastLED.addLeds<WS2812, D2, COLOR_ORDER>(leds, 256).setCorrection( TypicalLEDStrip );
  FastLED.addLeds<WS2812, D3, COLOR_ORDER>(leds, 256, 256).setCorrection( TypicalLEDStrip );
```

Это две матрицы 16x16 параллельно на D2 первая, на D3 вторая

vvip-68, 2023-02-04:
А уж "заставить один процессор переподключаться на 2 mac адреса" -вообще из области фантастики. Как вы это себе вообще представляете?

cochelevalexander, 2023-02-04:
Сначала на один, а после на другой, но вывод информации после загрузки всего массива

vvip-68, 2023-02-04:
Вы представляете сколько времени занимает подключение WiFi даже если это было бы вообще возможно такое переключение? В FPS уроните до 0.01.. ТО есть 1 кадр за 10 секунд. Я не представляю как это технически вообще реализуемо. Я думаю, что никак.

vvip-68, 2023-02-04:
Вон простейший поиск в гугле дает примеры использования ESP32 с параллельным выводом на 24 пина (24 отрезка ленты) с общим количеством пикселей 8800 с FPS равным 90. Изучайте пример, повторяйте...

cochelevalexander, 2023-02-04:
Мне нужно данные с Junix разбить на два пина. При подключении сетки 2х3 на одну ножку, проявляется накладка изображения ( Из-за применения 10 вселенных вместо 7). Другими словами, на один пин, для адекватной работы, можно ставить не более 4-х матриц (16х16), а их стоит 6. Сам Junix отправляет изображение с частотой в 25 fps, предельная для диодов около 20. Из-за этого часть данных просто отсеивается, и на сетке получается вместо адекватной картинки - полная каша.

vvip-68, 2023-02-05:
Я уже написал выше, какая константа отвечает за количество вселенных, чтобы было 10 вместо 7.
Думаю что основным тормозом, влияющим на FPS является все же процесс вывода на ленту, и прием 10 вселенных вместо 7 - не большая проблема и быстродействия хватит. Если что - ESP32 - куда более производительный микроконтроллер - используйте его.

Я уже написал выше как организовать "параллельный" вывод на ленты с двух-трех-четырех-сколько надо пинов.
Это самое простое без физической переделки. Сомневаюсь, что на одноядерном ESP8266 может быть реальная параллельность - не проверял, но кто мешает проверить то? 
ESP32 имеет два ядра. Тут уже возможна реальная параллельность вывода. Надо пробовать.

Возможно на "стандартной" библиотеке FastLED реальная параллельность и не работает, но простым поиском в гугле находится несколько клонов FastLED в которых реализована реальная параллельность вывода. Я уже писал выше, что люди на форумах пишут, что  добились FPS=90 на 8800 диодах.

В сети есть несколько примеров (и даже видео на ютуб) как делают параллельный вывод на ESP32 и нескольких регистрах сдвига до 100 параллельных лент с FPS=130.

Так что изучайте как делают люди, пробуйте, дописывайте или переписывайте скетч и всё у вас получится

cochelevalexander, 2023-11-11:
Здравствуйте, а есть ли возможность вставить в junix framer таблицу индексов сделанной с помощью LedMapper, вместе с встроенными в него способами подключения матрицы?

vvip-68, 2023-11-11:
Нет, нет такой возможности. Как-то эти две возможности развивались совершенно параллельно, независимо друг от друга и нигде не пересекались. Не реализовано. И вряд ли будет, потому как случай довольно специфичный, и я уже плохо помню что там и как для этого нужно переделывать.

Зато с прошивкой вот что: эффект SD-карта - вывод кадра изображения на матрицу выполняется тут (sdcard.ino)

[image]

Для экономии памяти - загрузка данных выполняется сразу в область памяти, назначенную как буфер светодиодов.
Если матрица небольшого размера (нужно мало памяти под буфер экрана) или используется ESP32, в котором много памяти - можно под загрузку кадра из файла выделить отдельный буфер, читать кадр в него, а потом из этого промежуточного буфера циклом переписывать данные с иcпользованием функции drawPixelXY() которая уже использует карту индексов для правильной адресации диодов с учетом их реального физического положения в массиве.

Как-то так. Такие соображения...

cochelevalexander, 2023-11-11:
Матрица из 9 панелей 16х16, вывод на 3 пина, каждый из которых отвечает за 3 панели. Работает все отлично, esp32 wrom. Есть желание перевести на wrover 16r8 для подключения по tmp2. Junix создает файл, который хорошо отображается на матрице с флешки, но иногда появляются артефакты, которые, хотелось бы, подкорректировать...

vvip-68, 2023-11-11:
Ну, вот как-то так. Не умеет JinxFramer работать с картами индексов. Увы... Он писался ДО того, как карты индексов были придуманы как таковые и реализованы в прошивке...

cochelevalexander, 2023-11-11:
Понял, большое спасибо)

cochelevalexander, 2023-12-13:
Большое спасибо за последнее обновление jinx Frame Viewer, очень помогло!