vvip-68/GyverPanelWiFi #275: кнопка, wifi и плейер

bdu1977, 2023-02-08:
собрал гирлянду по последней прошивке, все работает , за исключением некоторых мелочей.
1. прошивка по воздуху не проходит, выдает ошибку [ERROR]: No Answer
сам светильник подключен к роутеру. при первом подключении к роутеру был задан один IP, а в arduinoIDE - инструменты - порт, показывает другой IP. если в скетче делать такой же IP (#define DEFAULT_IP {192, 168, 1, 60}     // Сетевой адрес устройства по умолчанию) как и в arduinoIDE то выходит вот что Uploading............ точки идут бесконечно
2. кнопка стоит сенсорная, она работает только на вкл и выкл, а так же на изменение яркости, как больше так и меньше, переключение эффектов нету. так и должно быть?
3. player использую не оригинальный, после прошивки в мониторе порта выходит ошибка, что plaer не обнаружен. в программе на смартфоне, я могу выставить музыку будильника и рассвета, все проигрывается, если выбрать мелодию и нажать на воспроизведение, а вот если выставить режим будильника, то он срабатывает через раз.
Помогите разобраться.

vvip-68, 2023-02-08:
5 вольт работать не будет вовсе или будет нестабильно). Не должно быть рядом сильноточных 1. А устройство пингуется, например с компа? Телефоном к нему можно подключиться? В настройках роутера снята галочка "Изоляция клиентов" или что-то подобное? Никакой VPN не стоит, который перенаправляет пакеты в другую сеть?

2. Функционал сенсорной кнопки точно такой же как и у обычной тактовой. Только сенсорные кнопки - слишком капризные. Провод до нее должен быть короткий, желательно экранированный. Питание - с пина микроконтроллера 3.3 В (при питании от +5 вольт работать не  будет или нестабильно)ю Рядом не должно быть сильноточных цепей, дающих наводки. Рядом с кнопкой не должно быть никаких предметов, проводов и прочих вещей, на которые она реагирует. В логе (в мониторе порта) при срабатывании должно выводиться количество кликов, которые она распознала. Мне не удалось нормально победить кнопку - все что я пытался собирать - работало нестабильно. Я ее выкинул. Чего и вам советую.

3. Странно. Если плеер не обнаружен (как пишет в логе) - никаких элементов связанных с плеером в программе быть не должно. Никаких настроек выбора звука и все такое. Если есть- значит все-таки обнаружен. По даташиту питание плеера должно быть от 3.3 вольта, до 4.2 вольта. При превышении питающего напряжения может работать нестабильно. Поскольку мощность плеера на выходе - 3 вата, то при питании 3.3в-4.2в он будет потреблять ток от 1 до 1.5 ампер. Источник питания должен это обеспечивать. Желательно, чтобы по питанию так же было стабильное напряжение, без помех. Помехи по питанию монут сводить его с ума.  Сам по себе плеер достаточно капризен. Кому-то везет и он работает нормально, кому-то нет и он перебирает экземпляры плеера и/или настройки в программе -  например GUARD_DELAY. Что пишет в логе по поводу плеера? Как при старте, так и в момент срабатывания / не срабатывания будильника (то есть - включения звуков).

bdu1977, 2023-02-09:
я в этом плане полный чайник, для меня многие фразу непонятны, но в любом случае спасибо за ответы.
1. не знаю как его запинговать с компа, но в настройках роутера при подключении гирлянды появляется новое устройство, но с другим IP не тем который указываю в скетче. подключиться к гирлянде через телефон и спокойно могу, VPN нету. логи могу посмотреть только вечером после работы. главный вопрос этот, как заставить обновляться по wifi
2. про кнопку я понял, значит она работает как обычная. пробовал ее включать от 3.3В но она не запустилась, отказывается работать и все, даже не моргает. пришлось подключить от 5В и все три провода, как Гайвер советовал скрутив в косичку. сама кнопка находится под пластиком примерно5мм и около металлической рамки гирлянды, срабатывает нормально. может просто повезло. 
3. плеер так же от 3.3В не запустился, он при вкл.питания сразу начинает хрипеть, заработал от 5В. в логах пишет ошибка плейера, но при этом (в логах) находит файлы на карте памяти). в приложении я могу выбрать звук будильника и звук рассвета. но замечена вот еще какая штука. если в приложении выбрать звук будильника и нажать кнопку проиграть, то звука никакого не будет, сколько не нажимай. а если выбрать и проиграть звук рассвета, то потом и звук будильника можно проиграть любой. но сам будильник срабатывает через раз. т.е. выставил на 8.00 будильник, время прошло он не сработал, тут же переставляю на 8.05 - будильник срабатывает, следующий раз новое время, будильник не срабатывает. и тут не имеет значения включена гирлянда или нет. 
Вечером сделаю копию логов и пришлю

vvip-68, 2023-02-09:
1. Команда системы (windows / Linux) - ping <ip>
3. А где вы брали 3.3В для плеера? Если с пина 3.3D микроконтроллера - он не способен выдать требуемый ток для плеера (1-1.5) ампера и вовсе может сгореть. При питании плеера от напряжения более 4.2 вольта он может работать очень нестабильно. Обычно прокатывает схема с подпирающим диодом на землю так, что с 5.2-5.6 вольта питания ленты на микроконтроллер и плеер приходит на 0.6-0.8 вольт меньше (то есть 4.2-4.8 вольта) для их питания и они работают более-менее стабильно.

bdu1977, 2023-02-09:
1. контроллер пингуется с компьютера, но обновление не происходит по воздуху.
2. кнопа и плейер запитаны через отдельный преобразователь на 3.3В
наверное придется обойтись пока без обновление по воздуху и капризной работой плейера. в логе видно, что сам плейер не определился, но файлы будильника на карте памяти видны. так же я могу управлять и включать звук с телефона. в логе последние строки писались когда включал и выключал звук через телефон. надеюсь файл с логом прикрепился.
[лог.txt](https://github.com/vvip-68/GyverPanelWiFi/files/10700330/default.txt)

vvip-68, 2023-02-09:
Ну, в логе ничего подозрительного не вижу.
У меня больше нет идей.

По обновлению по воздуху - вы выбрали файловую систему микроконтроллера с поддержекой OTA? Ведь верно же?

bdu1977, 2023-02-13:
> По обновлению по воздуху - вы выбрали файловую систему микроконтроллера с поддержкой достаточно места под OTA? Ведь верно же? То есть ФС:2MB; ОТА (1019Кб)

да все верно. перепробовал все варианты, но результат один. буду прошивать как обычно.

closed 2023-02-13 by bdu1977