vvip-68/GyverPanelWiFi #278: Параллельное подключение линеек светодиодов работет, как зигзагообразное

ACKET2009, 2023-04-16:
Здравствуйте. Распаял светодиоды, которые идут блоком 10х10. У них направление на планках в одну сторону. Прошиваю со своими настройками подключения, горизонтально радуга идёт красиво, а вертикальная чресстрочно во встречных направлениях и никак не меняется. Попробовал через андроид приложение WiFi panel setup - ничего не меняется так же...Где-то алгоритм не правильно работает, либо не срабатывает совсем...

vvip-68, 2023-04-16:
Какую версию прошивки вы используете?
Исходя из репозитория - что-то из версий 1.13?
Я проверил - у меня работает правильно - по разному для типов матрицы параллельная или зигзаг.

Какое у вас подключение - угол и направление из угла?
Попробуйте обновиться с репозитория. Я сегодня выложил апдейты - переход для ESP32 на версию ядра 2.06.
Что пишет в мониторе порта в момент старта прошивки?

В версии прошивки 1.14 - действительно была ошибка конфигурирования зигзаг/параллельная, но...
во первых - ошибка не в прошивке, а в программе конфигурации 
во вторых - репозиторий с этой версией был открыт только вчера и нигде публично не еще анонсировался...

Так что - я в непонятках...

ACKET2009, 2023-04-18:
20:26:26.135 -> WiFiPanel v.1.13.2023.0416
20:26:26.135 -> Версия EEPROM: 0xFF
20:26:26.135 -> Обновлено до: 0x5D
20:26:26.135 -> Host: 'WiFiPanel'
20:26:26.135 -> 
20:26:26.135 -> Инициализация файловой системы... 
20:26:26.323 -> Использовано 16384 из 1024000 байт
20:26:26.323 -> 
20:26:26.323 -> Инициализация EEPROM...
20:26:26.323 -> Сохранение строк выполнено.
20:26:26.323 -> Использованы адреса EEPROM 1000 - 1037
20:26:26.323 -> Свободно ячеек 3058
20:26:26.370 -> Настройки сохранены в EEPROM
20:26:26.370 -> 
20:26:26.370 -> 
20:26:26.370 -> Матрица: плоская 10x10
20:26:26.417 -> Адресация: по подключению
20:26:26.417 ->   Угол: левый верхний
20:26:26.417 ->   Направление: вправо
20:26:26.417 ->   Тип: параллельная
20:26:26.417 -> 
20:26:26.417 -> Доступные возможности:
20:26:26.417 -> + Бегущая строка: шрифт 5x8
20:26:26.417 -> + Кнопка управления: тактовая
20:26:26.417 -> + Синхронизация времени с сервером NTP
20:26:26.417 -> - Управление питанием
20:26:26.417 -> + Получение информации о погоде
20:26:26.417 -> - Управление по каналу MQTT
20:26:26.464 -> - Групповая синхронизация по протоколу E1.31
20:26:26.464 -> - Дополнительный индикатор TM1637
20:26:26.464 -> - Эффекты Jinx! с SD-карты
20:26:26.464 -> - Поддержка MP3 Player
20:26:26.464 -> 
20:26:26.464 -> Поиск файлов карт индексов...
20:26:26.464 -> Нет файлов карт индексов
20:26:26.464 -> 
20:26:26.464 -> 
20:26:26.464 -> Найденные анимации:
20:26:26.464 ->    Сердце	8x8
20:26:26.464 -> 
20:26:26.464 -> Поиск слайдов на FS в папке '10p10'...
20:26:26.464 -> Поиск слайдов на FS в папке '16p16'...
20:26:26.464 -> Слайды не найдены. Будут использованы картинки эффекта 'Погода'
И при этом вывод идёт как зигзаг... то есть бегущая строка чресстрочно бежит навстречу и радуга вертикальная так же... в общем работает не правильно. Об этом уже писали... Я не первый с этой проблемой...

ACKET2009, 2023-04-18:
То есть, судя по выводу - всё выбрано правильно, пишет, что параллельное направление, но вывод идёт как зигзаг...

ACKET2009, 2023-04-18:
Ничего не поменялось... может проблема с размерностью 10х10?

vvip-68, 2023-04-19:
Там нет зависимости от размерности. В любом случае - поставь в настройках большую размерность, скажем 20x20 - посмотри как изменится поведение.

Я не видел, чтоб кто-то еще писал про такое поведение.
Параллельной матрицы у меня нет, но моя матриц, которая зигзагом, при включении в настройках "параллельная" начинает вести себя вот так, как описано - текст строчки начинают бежать навстречу друг другу, из чего я делаю вывод, что всё работает правильно.

Я не могу исправить ошибки которые не могу воспроизвести...

Могу в данном случае посоветовать следующее: в файле utility.ino в строке 288 есть функция getPixelNumber() которая переводит координаты X,Y в номер светодиода в цепочке. Перепишите ее в соответствии с вашей конструкцией.

Второй вариант - создать для вашей матрицы карту индексов, по которой та же функция переводит координаты X,Y в номер светодиода в цепочке. О том, как создавать карты индексов расписано тут 
https://github.com/vvip-68/LedPanelWiFi/wiki/Сборная-матрица,-состоящая-из-матриц-разного-типа-и-размера

vvip-68, 2023-04-20:
Мне приходят уведомления обо всех сообщениях, что пишут здесь на гитхабе (issues, discussions) и на форуме у гайвера в ветке про эту прошивку. Я не припомню сообщений об этой проблеме.

Если вы всё делаете правильно и у вас не работает, тогда что я делаю неправильно, что у меня все работает?
Буду благодарен, если вы локализуете кусок в коде, и укажете что в нем неправильно и как исправить, потому что я данную проблему воспроизвести не могу, соответственно что-то исправить тоже.

Что значит "чем допустимо" более свежая версия? Я всё проверял после открытия вашей проблемы на самой последней версии, что в хранилище.  Алгоритмы вычисления номера диода по координатам от контроллера не зависят.

vvip-68, 2023-04-20:
Пришлите архив с вашим кодом (с вашими настройками) - куда-нибудь на файлообменник, тот вариант который не работает, я попробую скомпилировать его и посмотреть чем отличается

ACKET2009, 2023-04-20:
Хорошо, в субботу отправлю

ACKET2009, 2023-04-22:
https://disk.yandex.ru/d/Sx-thiarvM7Cfw вот сюда запаковал для 10х10

vvip-68, 2023-04-22:
Попробуй загрузить последнюю из хранилища.
Я помню, что эту ошибку исправлял, но оказалось, что исправлено на 1.14, в 1.13 - оставалась
А у меня не проявлялась, поскольку сбрасывалось в зигзаг, если файл не найден, а у меня он всегда присутствует. У тебя для твоего размера матрицы его нет. Поэтому сбрасывалось в Зигзаг

vvip-68, 2023-04-22:
По предыдущему комменту.
Настройки подключения к сети хранятся в файловой системе микроконтроллера.
Если прошивается с включением "Eras Flash" - "All Flash Content" - оно при прошивке стирает все, в т.ч и файловую систему. То есть - и настройки подключения к сети.

То же самое с загрузкой папки data в FS микроконтроллера - оно стирает при загрузке всю файловую систему. То есть - и настройки подключения к сети.

Файлы out в папке 16х16 - файлы роликов Jinx для матрицы 16х16 для эффекта SD-карта, когда на самом деле SD-карты нет, а включена опция эмуляции SD-карты файловой системой микроконтроллера

vvip-68, 2023-04-22:
> Картинки показывать начинает, но от роутера отваливается, при попытке подключить выводит строку, что нет доступа к сети wi-fi

Бывает, что после записи новых настроек в EEPROM и вызова команды Restart (программная перезагрузка) микроконтроллер не всегда может подключиться к сети. Я не знаю что там у него внутри происходит при этом. Достаточно нажать кнопку Reset - жесткая хардверная перезагрузка микроконтроллера - и он нормально подключается к сети с сохраненными настройками

ACKET2009, 2023-04-22:
> Попробуй загрузить последнюю из хранилища. 
Там сейчас один файл изменённый только - корневой скетч - его у себя поправить нужно?

P/S/ Ещё у меня была перед новым годом проблема с матрицей - после включения чуть-чуть работала а потом плавно затухала... Я как-то исправил - нашёл из за чего, а теперь при заливке новой, ввожу свои правки, а этот эффект опять проявился, а что исправлял - не могу вспомнить никак... старею, блин, память уже не та...

vvip-68, 2023-04-22:
Конкретно - вот это исправление. В одном файле
[image]

Яркость автоматически уменьшается в ноль, если в настройке стоит, что подключена сенсорная кнопка на d4 , а на самом деле физически она не припаяна Вариант (похоже твое исправление) - переназначение кнопки на D6

Либо когда вместо ядра 2.7.4 ставится ядро из серии 3.x.x - тогда вот такой эффект + нестабильная работа вообще в целом

vvip-68, 2023-04-22:
Там в комплекте есть файл-пример public.zip - в нем отдельные файлы, не включенные в проект. В них хранятся гастройки "железа" (что куда подключено) и пароли. То есть, т.к. их нет в проекте - они не обновляются с очередным обновлением скетчей, а остаются с твоими настройками. При компиляции подхватываются именно они.  То есть не надо заново перенастраивать параметры в a_def_hard.h и в a_def_soft.h

Включается использование этих файлов тут:
[image]

Настоятельно рекомендую использовать этот вариант хранения своих настроек, чтобы каждый раз не вспоминать где что менял.

ACKET2009, 2023-04-22:
Вот спасибо, а то я закомментировал эти строки, чтобы ошибки не лезли, а чего и зачем и откуда берутся - не разобрался!

closed 2023-04-23 by vvip-68