vvip-68/GyverPanelWiFi #281: бегущая строка

bdu1977, 2023-11-01:
создал несколько бегущих строк. из приложения все строчки при нажатии на кнопку просмотр, появляются, все четко. но при работе самой гирлянды, на экран выводятся только простые строки без макросов или с макросами, но не временными. в настройках поставил смену строк через 20сек, перед строками которые появлялись, в приложении ставил {-} и сохранял, в итоге остались только те где были временные макросы, и эти строки не выводились на матрицу, сколько бы я не ждал. при чем показ часов и погоды и температуры, это работает вот пример этих матриц, может что-то не так писал:
до нового года осталось {R01.01.***+}{S01.10.****#31.12.****23:55:59}
{B#000000}Пора спать, спокойной ночи.{S23:30#23:59}
{B#000000}До{C#00D0FF}Рождества{R07.01.****}осталось{S05.01.****#06.01.****}

vvip-68, 2023-11-02:
Проверил у себя.
Вариант 1 (про Новый год) - у меня отображается с заданной периодичностью. Единственное, что между датой и временем '#31.12.**** 23:55:59' по спецификации нужен пробел, но у меня показывается и без него. Но попробуйте все же добавьте.

По второму варианту - макрос {S} вроде как требует присутствия даты. Иначе он время (без даты) пытается интерпретировать как дату и у него ничего не получается. В варианте 

> '{B#000000}Пора спать, спокойной ночи.{S**.**.**** 23:30#**.**.**** 23:59}' 

у меня макрос работает.

Третий макрос - аналогичен первому. Если указать правильные даты интервала показа - он отображается. Если же как указано - с 05.01 до 06.01 - ну так он СЕЙЧАС показываться и не должен. Вот придет 5 января...

Больше ничем помочь не могу...

bdu1977, 2023-11-02:
получилось, вопрос закрыт. спасибо.

closed 2023-11-02 by bdu1977