vvip-68/GyverPanelWiFi #282: Ошибка компиляции

Inthestep, 2023-11-01:
Arduino: 1.8.19 (Windows 10), Плата:"LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Inthestep, 2023-11-01:
как скачать библиотеки 
hardwareserial и ArduinoJson
ПАМАГИТЕ!!!

vvip-68, 2023-11-02:
Библиотеки нужно брать из папки проекта, так как скачанные из интернета могут иметь версии в которых изменено настолько, что проект не будет собираться.

hardwareserial - это не отдельная библиотека, это часть ядра микроконтроллера и идет вместе с ним.
ArduinoJson - находится в папке проекта 'libraries'

Inthestep, 2023-11-02:
Спасибо!

closed 2023-11-02 by Inthestep