vvip-68/GyverPanelWiFi #289: Проблема при добавлении грузинского языка.

CrazyLazyMan, 2023-12-16:
Здравствуйте! Столкнулся с проблемой при добавлении грузинского языка, само собой все 33 символа отрисовать и добавил в массив fontHEX в файле fonts.h, но не знаю как отремонтировать getFont в runningText.ino т.к. грузинские символы в utf-8 кодируются тремя байтами. Второй вопрос, не надо ли как то переделывать андроид приложение, т.к. после отправки грузинских символов на матрицу грузинский символ превращается в вопросительный вопрос в ромбе(см. скриншот). ESP32, матрица 16х16.
[image]
Буду признателен за помощь.

vvip-68, 2023-12-16:
Ну это вам много чего переписывать нужно. И сам алгоритм в getFont() для трех-байтового кодирования м много еще для чего. И не уверен, что приложение вообще поддерживает трехбайтовые символы. Так то там UTF-8, но, как говорится, есть нюансы. Всё это ох, как непросто будет... Не хотите попробовать версию 1.14 из соседнего проекта. Там по крайней мере все настройки в браузере - он то точно грузинский должен поддерживать. Тогда останется только перенести отрисованные символы в тот проект и переписать функцию получения шрифта и вывод на матрицу в скетче...

CrazyLazyMan, 2023-12-16:
Спасибо за столь быстрый ответ. Мои худшие опасения подтвердились, с другой стороны, наверное все же перейду на 1.14.

closed 2023-12-16 by CrazyLazyMan

vvip-68, 2023-12-16:
Поздновато вы к Новому году готовиться начали... :)

CrazyLazyMan, 2023-12-16:
У меня матрица на рюкзаке)