vvip-68/GyverPanelWiFi #293: помогите скомпилировать для ESP32 пожалуйста

IgorBratsk, 2023-12-20:
не получается(((

Arduino: 1.8.13 (Windows 7), Плата:"ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Igor\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Igor\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Igor\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Temp\arduino_build_588777 -warnings=none -build-cache C:\Temp\arduino_cache_882896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.mklittlefs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.openocd-esp32.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230419 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230419.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230419 -verbose C:\Users\Igor\Downloads\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.13\GyverPanelWiFi_v1.13.ino

C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Igor\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Igor\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Igor\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Temp\arduino_build_588777 -warnings=none -build-cache C:\Temp\arduino_cache_882896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.mklittlefs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.openocd-esp32.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230419 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230419.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230419 -verbose C:\Users\Igor\Downloads\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.13\GyverPanelWiFi_v1.13.ino

[... cut ...] 3.ino.bootloader.bin" "C:\\Users\\Igor\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.14\\tools\\sdk\\esp32\\bin\\bootloader_qio_80m.elf" ) )

����� "C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1\esptool.exe" �� ᮢ���⨬� � ���ᨥ� Windows, ࠡ���饩 �� �⮬ ��������. �஢���� ᢥ����� � ��⥬�, �⮡� 㧭���, ����� ����� �ணࠬ�� x86 (32-ࠧ�來��) ��� x64 (64-ࠧ�來��) ��� �㦭�, � ������ �� � ���⠢騪� �ணࠬ����� ���ᯥ祭��.

exit status 1

Ошибка компиляции для платы ESP32 Dev Module.

vvip-68, 2023-12-20:
А кто вам сказал, что версия 1.13 будет вообще компилиться под ядро 2.0.14?

IgorBratsk, 2023-12-20:
Я дико извиняюсь, к сожалению сложнее арлуины скетчи не заливал. А можно это ядро понизить как то?

IgorBratsk, 2023-12-20:
Спасибо огромное, что помогаете, завтра попробую так сделать. А то что у меня версия ide не такая может влиять?

vvip-68, 2023-12-20:
В данном конкретном случае - думаю нет. Но все же лучше поставить 1.8.19

IgorBratsk, 2023-12-21:
Arduino: 1.8.19 (Windows 7), Плата:"ESP32 Dev Module, FTDI Adapter, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Igor\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Igor\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Igor\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=external,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Temp\arduino_build_441872 -warnings=none -build-cache C:\Temp\arduino_cache_245986 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.openocd-esp32.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20220706 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20220706.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20220706 -prefs=runtime.tools.mklittlefs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.mkspiffs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1 -prefs=runtime.tools.esptool_py-4.2.1.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -verbose C:\Users\Igor\Downloads\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.13\GyverPanelWiFi_v1.13.ino

C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Igor\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Igor\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Igor\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=external,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Temp\arduino_build_441872 -warnings=none -build-cache C:\Temp\arduino_cache_245986 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.openocd-esp32.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20220706 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20220706.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20220706 -prefs=runtime.tools.mklittlefs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=C:\Users\Igor\AppData\Local\ [... cut ...] ages\\esp32\\hardware\\esp32\\2.0.6\\tools\\sdk\\esp32\\bin\\bootloader_qio_80m.elf" ) )

����� "C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1\esptool.exe" �� ᮢ���⨬� � ���ᨥ� Windows, ࠡ���饩 �� �⮬ ��������. �஢���� ᢥ����� � ��⥬�, �⮡� 㧭���, ����� ����� �ணࠬ�� x86 (32-ࠧ�來��) ��� x64 (64-ࠧ�來��) ��� �㦭�, � ������ �� � ���⠢騪� �ணࠬ����� ���ᯥ祭��.

exit status 1

Ошибка компиляции для платы ESP32 Dev Module.

все таки видимо нужно взять и все установить сначала на W10 64 bit :((

vvip-68, 2023-12-21:
Про 64 бит не знаю - вряд ли это что-то изменит. 
Я в этом тексте даже ни одной ошибки внятной не вижу...
А если попробовать с ядром 2.0.14? Я не пробовал, но вдруг...

vvip-68, 2023-12-21:
Ошибка то не компиляции, ошибка загрузки скомпилированного бинарника в плату, похоже...

IgorBratsk, 2023-12-21:
Ну вот, там среди этих ироглифов в конце то 32 то 64, видимо пишет, что версия этого есптул для 64 битного виндовса. Ладно, попробую с нуля все как надо сделать.

vvip-68, 2023-12-21:
А почему вообще иероглифы? У меня их нет...

IgorBratsk, 2023-12-21:
Ну видимо версии чего то с чем то не совпадают. Не может ide отобразить ошибку что-ли.

closed 2024-01-24 by vvip-68