vvip-68/GyverPanelWiFi #295: Лагание бегущей строки и подвисание после долгой работы.

CrazyGhostRider, 2023-12-22:
Здравствуйте.
1) Заметил. что бегущая строка чуть подвисает каждые 3-4 сек.  Но может и идеально работать.
2) Настроил, чтобы отключался демо режим на светлое время дня.  Но на четвертый день непрерывной работы демо режим включился с производительностью 1-2 fps. (Включилась бегущая строка, но двигалась со скоростью 1 столбец матрицы в 5 секунд) 

Проблема 1 и 2 происходит одинаково на двух независимых гирляндах. 

ESP 32
WiFiPanel v.1.13.2023.1205
Версия ядра ESP32   - 2.0.6
Версия FastLED      - 3.5.0
Arduino IDE - 1.8.19

vvip-68, 2023-12-22:
Причина неизвестна. Нужно смотреть что в логах в это время происходит. Без этого - никак.
Может быть память фрагментируется (хотя в этой версии прошивки не было замечено).
Может теряет соединение с интернетом и все паузы - на ожидание ответа из сети.
Пока ждет ответа на запрос - ничего другого делать не можкт.
В любом случае - без лога, выводимого в монитор порта ничего не скажешь...
Да и с ним возможно тоже.

CrazyGhostRider, 2023-12-23:
Подожду 4 дня и попробую через интернет подключиться к esp.
Мониторинг порта не получится сделать т.к. нет ноутбука.

vvip-68, 2023-12-23:
А вы роутер перезагрузить не пробовали? 
Я вот два дня промучился, что у меня перестала работать групповая трансляция. И так и этак. Ну ни в какую...
Потом перезагрузил роутер - и чудо! Все залетало...
Согласен, может не совсем тут случай, но вдруг все же во время этих тормозов устройство лезет в сеть за текущим временем или погодой, а роутер сеть для устройства заблокировал... И оно мучается не может...
А если роутер не перезагружать, но отключить получение погоды с интернета и синхронизацию времени?

CrazyGhostRider, 2023-12-23:
Первую проблему это не решает.

 А со второй: у меня роутер автоматически перезагружается. И нужно ждать ~4 дня.

CrazyGhostRider, 2023-12-29:
Итак по проблемам:
1) Любая бегущая строка всё также подлагивает (заедает)
2) Ардуинка не умеет переподключаться к wi-fi при перезагрузке роутера (Asus RT-AX58U) и зависает.
В роутере МАС адреса ESP32 прописаны. Не думаю, что роутер блокирует интернет. И старый wi-fi принтер без проблем подключается к сети.

П.с. Идея по улучшению: Можно добавить датчик освещенности. И им регулировать яркость светодиодов. (Например: Висит матрица на стене и днем ярко горит, а ночь еле светит, чтобы не мешать спать) Либо вкл/выкл демо режим и т.п.

vvip-68, 2023-12-29:
1. Бегущая строка. Сдвиг происходит по таймеру сдвига, задающему скорость бегущей строки.
Контроллер кроме собственно рисования занимается еще каким-то своим делами. Например проверяет погоду, время, анализирует кнопки, ядро что-то свое делает. Если поставить в цикл перед каждым вызовом отрисовки матрицы сколько прошло времени с предыдущей отрисовки  - циферки бегут примерно следующие 20б 21,25,20,**48**,30, 20, 22, 24, **50**,20,23 и так далее. То есть в какие-то моменты ядро / прошивка делает что-то свое и вызовы происходят неравномерно. На эффектах это не особо заметно. На бегущей строке воспринимается как неравномерное движение и подергивание.

Есть идеи как это победить? Я готов выслушать и посмотреть что можно сделать.

2. Переподключение. Везде во всех мануалах пишут что ESPшка делает это автоматически. У меня не было основания не верить мануалам. Поэтому особо не проверял. Просто помнится, что если такое и случалось - то проблем это не вызывало. Позже посмотрю, проверю.

3. Датчик освещенности уже был. Показал свою полную неадекватность в том смысле, что реальные жизненные сценарии не соответствуют такой прямолинейной зависимости - горит свет включили ярко, стемнело - светим менее ярко. В жизни всё гораздо сложнее. Поэтому был выкинут за неприменимость в реальной жизни. И возвращен не будет.

vvip-68, 2023-12-29:
Про рисование текста бегущей строки

Мы проходим цикл. На один цикл уходит 20-25 мс. 
В это время проверяется таймер эффекта - нужно ли перерисовывать очередной кадр эффекта (зависит от выставленной скорости эффекта). Если нужно - запускается алгоритм рисования. Вот он занимает примерно 20 мс на формирование кадра.

Кадр нарисовали. Проверяется нужно ли сдвигать текст. А до срабатывания таймера сдвига еще 5 мс осталось. Текст не сдвигаем в этом цикле.

Следующий оборот таймера. Ок Время пришло - рисуем новый кадр (20 мс). Но тут и таймер сдвига подоспел. Только он (сдвиг) уже просрочен на 20 мс...

В результате наблюдаем примерно следующее. 
Здесь Shift - выполнение сдвига текста. Циферки - отрисовка каждого кадра (сколько миллисекунд прошло с предыдущего вывода на матрицу)

[image]

То есть из за несовпадения таймеров отрисовки кадра эффекта и сдвига текста - иногда сдвиг происходит на каждый кадр, иногда через кадр. Все это - как мелкие подергивания.

А бывает совпадение настроек скоростей, когда два сдвига , а вывода не было - тогда при следующем выводе кадра на матрицу текст "прыгнет" сразу на две колонки.

И вот что со всем этим делать - я не знаю...

CrazyGhostRider, 2023-12-29:
А не менялась ли работа с бегущей строкой с прошивки "WiFiPanel v.1.13.2022.1212"? Т.к. в прошлом году я только её использовал. (Если помните, то более новые прошивки у меня давали сильные помехи) И если бы были проблемы с бегущей строкой, то сообщил бы ранее.

vvip-68, 2023-12-29:
Вопрос непонятен.
В Версии 1.13 в бегущих строках изменена работа макроса {P} - добавлена возможность вывода обратного отсчета по центру, изменена работа макроса {D} - добавлен спецификатор вывода 'yyy+' - следующий шод.

В самом алгоритме когда рисовать эффект / сдвигать строку - ничего не менялось.
В Версии 1.13 рекомендовано использование ядра 2.7.4 и FastLED 3.4. Вот с этим подергивания заметно меньше ощущается. Если 1.13 собрать под ядро 3.1.2 и FastLED 3.6 - оно соберется, ничего не меняя в скетче. Но хдесь подергивания текста гораздо заметнее.

В текущей 1.14 ветки main - пвбота с бегущей строкой точно такая же как в версии 1.13. Но ядро нужно 3.1.2 (под 2.7.4 кажется даже не соберется).

В 1.14 из соседнего брэнча - там пытался крутить и так и сяк, всякие FASTLED_ALLOW_INTERRUPTS включал/выключал. refreshRate пытался подобрать. Задержки ставил и убирал. Там может как-то по другому всё это работать.

Вот например кто-то тут или на форуме писал - в какой-то момент вывод замирает. На матрице ничего не меняется, хотя управление доступно. Словил такой момент. Ранее не было. Картинка статична, только самый первый диод мерцает. Оказалось если компилить без FASTLED_ALLOW_INTERRUPTS - происходят вот такие замирания. А с ним - не происходит. Хотя и это все спонтанно / спорадически и непредсказуемо как-то случается. Зато с этим параметром ингода стало крашится с сообщением "Exception 4: Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register"

И я не понимаю что с этим делать...

vvip-68, 2023-12-30:
Про подергивания бегущей строки и в целом...
Два дня мучаюсь, перебирая разные варианты как на esp8266 так и esp32
Ядро для 8266 - 3.1.2, для ESP32 - 2.0.14

На ESP32 более-менее нормально работает с FastLED 3.6 без дефайнов, о которы[ ниже
На ESP8266 наиболее мягкий скроллинг текста и эффектов на FastLED 3.5 и с дефайнами.

Попробуйте подбирать варианты - может какая-то комбинация приведет к успеху.
У меня впечатление, что это зависит от фаз луны. То что вчера работало сегодня отказывается работать напрочь.

Брал не изменяя один и тот же вариант скетча. По контрольному выводу в порт - на каждый FastLED.show() - 
вывод идет регулярно, с почти равными промежутками.

На ESP8266 - перебирал версии FastLED 3.4, 3.5, 3.6 и комбинации вкл/выкл прерываний
На каких-то вариантах горит только первый диод, на матрице ничего. 
На каких-то - есть вывод на матрицу, первый диод тоже мерцает. Иногда вывод застывает, первый диод продолжает мерцать.
На каких-то вариантах диод не мерцает, вывод на матрицу идет, но текст с сильными рывками.
И это все не меняя ни строчки кода.

Пока наиболее стабильно ваниант с FastLED 3.5 и в инклудах

#if defined(ESP8266)
  #define FASTLED_ALLOW_INTERRUPTS 0
  #define FASTLED_INTERRUPT_RETRY_COUNT 1
#endif
#include <FastLED.h>

Пробуйте, может у вас какой-нибудь тз вариантов взлетит...

vvip-68, 2023-12-30:
Сейчас в верке версии 1.14 ntp_lwip выложена последняя и на мой взгляд наиболее стабильная версия. У меня на окнах сейчас стоит она. В локальном режиме подергиваний бегущей строки вообще не наблюдаю.
Ядро 3.1.2, FastLED 3.5,  FASTLED_ALLOW_INTERRUPTS 0

CrazyGhostRider, 2023-12-30:
Хорошо, попробую.
Но у меня есть только ESP32

vvip-68, 2023-12-30:
Ну на ESP32 у меня в этой системе на этой версии скетча работает сам контроллер - который основной и ведет трансляцию на три окна. Для него все то же только версия ядра 2.0.14

CrazyGhostRider, 2024-01-02:
Скомпилил с ядром 2.0.14 и FastLED 3.5. А потом с FastLED 3.6 — разницы не заметил по сравнению со стандартными настройками. (Ядро 2.0.6 и FastLED 3.5)
Но теперь у меня одно окно на 2.0.6, а другое 2.0.14 работают.

MaxWh1t3, 2024-01-04:
у меня лагает, если включить отображение погоды, НО у меня и частота проа стоит 80, при 160 не компилировалось...

closed 2024-01-24 by vvip-68