vvip-68/GyverPanelWiFi #296: не правильная кдировка цветов

Kenaed, 2023-12-23:
Купил готовую гирлянду в ней другая кодировка RGB можно ли это исправить в коде? и думаю стоит такую возможность включить в настройки

vvip-68, 2023-12-23:
Найти константу COLOR_ORDER и там исправить порядок цветов
Включит в настройки - технически сложно.

frol-aleksan, 2024-01-09:
Включить в настройки - технически сложно.

Ничего сложного. Сделал у себя за пару часов.

vvip-68, 2024-01-09:
Тогда научите как

frol-aleksan, 2024-01-09:
Все делал по аналогии с уже имеющимся функционалом.
1. Начинаем с того, что в приложении Panel Setup добавляем комбобокс cbColorOrder со списком возможных вариантов порядка цветов (GRB,RGB,BGR,BRG,GBR,RBG); из них у меня лично используются GRB, RGB и BGR.
2. На экране редактирования блоков добавляем функцию ParamSetM10 и вставляем ее выполнение в paramSwitch6
[image]
3. Ковыряем код прошивки (буду указывать, что в каком файле делал)
 Заглавный файл
 где настройки ленты
 switch(mCOLOR_ORDER) {
    case 0: FastLED.addLeds<WS2812, LED_PIN, GRB>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip ); break;
    case 1: FastLED.addLeds<WS2812, LED_PIN, RGB>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip ); break;
    case 2: FastLED.addLeds<WS2812, LED_PIN, BGR>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip ); break;
    case 3: FastLED.addLeds<WS2812, LED_PIN, BRG>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip ); break;
    case 4: FastLED.addLeds<WS2812, LED_PIN, GBR>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip ); break;
    case 5: FastLED.addLeds<WS2812, LED_PIN, RBG>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip ); break;
  }

a_def_soft.h
где объявляются переменные ширины и высоты матрицы
uint8_t    mCOLOR_ORDER = 0;                     // порядок цветов в гирлянде, 0 соответствует GRB

a_main.ino
где case 8: (настройки параметров матрицы)
case 8:  // Настройки параметтров матрицы (размеры, подключение, порядок цветов)
      str = getStateString("UP|FM|M0|M1|M2|M3|M4|M5|M6|M7|M8|M9|M10"); 
      break;

Между if (key == "M9") {...} и if (key == "E0") {...} добавляем
// порядок цветов в гирлянде
  if (key == "M10") {
    tmp = String(mCOLOR_ORDER);
    if (value) {
      value->set(tmp);
      return tmp;
    }
    return str + "M10:" + tmp;
  }

eeprom.ino
где идет определение направления и точки подключения
mCOLOR_ORDER      = getColorOrder();

putColorOrder(mCOLOR_ORDER);

void putColorOrder(uint8_t colorder) {
  if (colorder > 5) colorder = 0;
  if (colorder != getColorOrder()) {
    EEPROMwrite(93, colorder);
  }  
}
uint8_t getColorOrder() {
  uint8_t value = EEPROMread(93);
  if (value > 5) value = 0;
  return value;
}

mqtt.ino
где #define STATE_KEYS
#define STATE_KEYS "|W|H|DM|PS|PD|IT|AL|RM|PW|BR|WU|WT|WR|WS|WC|WN|WZ|EF|EN|UE|UT|UC|SE|SS|SQ|BE|CE|CC|CO|CK|NC|SC|C1|DC|DD|DI|NP|NT|NZ|NS|DW|OF|TM|AW|AT|AD|AE|MX|MU|MD|MV|MA|MB|MP|AU|AN|NW|IP|QZ|QA|QP|QS|QU|QR|TE|TI|TS|CT|C2|OM|ST|AM1T|AM1A|AM2T|AM2A|AM3T|AM3A|AM4T|AM4A|AM5A|AM6A|UP|FM|T1|T2|M0|M1|M2|M3|M4|M5|M6|M7|M8|M9|M10/E1/E2/E3"

Может где-то что-то пропустил, ссылка на коммит с данными правками: https://github.com/frol-aleksan/GyverPanelWiFi/commit/d6af36948e548919e900d651471384f8da9188c4

Также там добавлен эффект для проверки правильности выбора порядка цветов, заполняющий верхние 3 ряда матрицы красным, зеленым и синим, кнопка для его запуска находится в основном приложении.

frol-aleksan, 2024-01-09:
Но пин один раз припаял, и все. А гирлянды можно подкинуть разные, и чтобы под каждую из них каждый раз не перепрошивать плату, добавил этот функционал

vvip-68, 2024-01-09:
Ну мне чаще попадались варианты (от пользователей), когда - подключил, а оно не работает и не знаю почему - как бы повыбирать на какое пин реально вывод идет. А с цветовым порядком - тот же аргумент - один раз в скетче прописал, проверил и больше менять не нужно. Мне как-то за эти годы не нужно было подкидывать разные гирлянды. Исходя из этого (из моих потребностей) и писалось.

Другой вопрос в последнее время всплыл, что мне китаец прислал три куска гирлянды - и у них у всех разный цветовой порядок. И тут уже нужно придумывать настройку - что диоды 0-99 - GRB, 100-199 - BGR и 200-299 - GBR, чтобы соединить это все в одну гирлянду

Но это уже совсем другая история...

closed 2024-01-24 by vvip-68

NLObP, 2024-12-19:
> Другой вопрос в последнее время всплыл, что мне китаец прислал три куска гирлянды - и у них у всех разный цветовой порядок. И тут уже нужно придумывать настройку - что диоды 0-99 - GRB, 100-199 - BGR и 200-299 - GBR, чтобы соединить это все в одну гирлянду
> 
> Но это уже совсем другая история...

Подскажите, как сделать? Такая-же беда! Китаец прислал две гирлянды с разным цветовым порядком.

vvip-68, 2024-12-19:
Все уже придумано, описано и рассказано давно
https://github.com/vvip-68/LedPanelWiFi/wiki/FAQ#в-порядок-цветов---две-разных-ленты-с-разным-порядком-цветов

vvip-68, 2024-12-19:
Видимо, все же придется в настройки в веб это выносить. Потому что каждый год одно и то же. Но пока - только вот так...