vvip-68/GyverPanelWiFi #297: Ошибки и непонятки

Cyxapuk11, 2024-01-02:
Не понимаю чего он от меня ждет (МК). Загрузил скетч для лед-панели, загрузка успешная, но по факту ничего не работает. Использую esp8266, пробовал по описанию разные варианты ядра(?), но результат примерно один и тот же. На панели горят пара светодиодов в углу, раздачи Wi-Fi нет. 
Поначалу, когда пробовал ядро 3.1.2, иногда получалось дождаться воспроизведения случайного эффекта, но через пару секунд все гаснет (читал об этой проблеме, но как решить не понял). Когда перешел на 2.7.4 вообще все затихло. 
Есть подозрение, что я не так или не те прикрепляю библиотеки, но какие и как правильно не чаю. 
Моя esp8266 куплена на али лет 5 назад, до этого она вполне работала и даже выдавала wifi, хотя в прошивке гайвера он не был задействован.
Пробовал разные пины (d2 d4) и менял местами их в прошивке. Пробовал регулировать напряжение 4.6-5.1В. Пробовал переподключить библиотеки, но те, что нужно было брать в файле сыпали ошибками при компиляции. Сейчас я уже не понимаю, скачанные менеджером это или из архива. 
В общем понимаю, что с такими проблемами и уровнем знаний меня проще послать колоть дрова да косить снег, но надежда помирает последней. Заранее спасибо за помощь

vvip-68, 2024-01-03:
Ну в вашем случае проще всего все удалить и поставить заново, сделать всё по инструкции.
IDE брать 1.8.19

https://github.com/vvip-68/GyverPanelWiFi/wiki/Пошаговая-инструкция-подготовки-среды-для-проекта
https://github.com/vvip-68/GyverPanelWiFi/wiki/Всё-собрал,-ничего-не-работает.
https://github.com/vvip-68/GyverPanelWiFi/wiki/Настройка-скетча-для-вашего-устройства

Cyxapuk11, 2024-01-03:
> Ну в вашем случае проще всего все удалить и поставить заново, сделать всё по инструкции. IDE брать 1.8.19
Переустановил IDE (как и был, на 1.8.19). Почистил "All flash contents", библиотеки не подключал вовсе, просто закинул в папку Arduino/lib. 
Картинка есть, точка доступа появилась, но называется PanelAP и запаролена. Как понимаю, по умолчанию зовется WiFiPanel и не запаролена? Или как подключаться к ней?

Cyxapuk11, 2024-01-03:
> Ну в вашем случае проще всего все удалить и поставить заново, сделать всё по инструкции. IDE брать 1.8.19
Переустановил IDE той же 1.8.19, почистил память контроллеру и не добавлял библиотеки в скетч. 
Сейчас wi-fi появился, но приложение к нему не подключается. Телефон подключился с паролем 12341234, но приложение не хочет. Пробовал перезагрузить МК, переустановить приложение.

vvip-68, 2024-01-03:
[image]

[image]

vvip-68, 2024-01-03:
Отключите на телефоне "Мобильные данные"

Cyxapuk11, 2024-01-03:
Я не понимаю порядок подключений. Почитал FAQ и запутался только больше. МК подключается к роутеру И создает свою точку сразу? Или создает точку для связи с приложением, получает из приложения настройки и данные о роутере, и далее, при надобности переподключается к роутеру? 
Мне нужно было вводить данные о роутере во вкладке a_def_soft? Локальная сеть это мой роутер или МК? 

Мобильные данные не помогли, адрес моего роутера, как я понял, 192.168.100.1. Его сюда вставить надо?
[image]

В порт МК шлет подобного рода шифры (матрица сейчас воспроизводит эффекты, точка доступа есть):
[image]

vvip-68, 2024-01-03:
Подключается ли матрица к роутере или нет - зависит от того указали ли вы в скетче (правильные) настройки сети или нет.
Если не указали - матрица не сможет подключиться к роутеру и создаст свою точку доступа.
Если матрица создала свою точку доступа и вы подключились к ней - можете из приложения настроить подключение матрицы к роутеру, задав имя сети и пароль к ней, а также IP адрес, который должно получить устройство после перезагрузки. Это самый простой путь, чтобы ]c его не прописывать в скетче.

Cyxapuk11, 2024-01-03:
В скетче такого не видел и не менял, в настройках вижу только Upload speed из подходящего: 
[image]

vvip-68, 2024-01-03:
[image]

Cyxapuk11, 2024-01-03:
Не помогло (настроил это и перезагрузил мк):
[image]

vvip-68, 2024-01-03:
Значит у вас микроконтроллер не работает. Хотя, говорте, что вывод на матрицу есть...
Я с таким не сталкивался, не подскажу.

Cyxapuk11, 2024-01-03:
Попробовал наугад остальные и нашел. Клал он на наши частоты, куда хочет - туда и шлет: 
[image]

vvip-68, 2024-01-03:
Это скорость вывода при старте микроконтроллера, на которой он выводит начальные свои условия старта. Больше на этой скорости ничего не будет. Весь вывод из скетча идет на 115200
Если вывода не 115200 нет - всего два варианта - или микроконтроллер не стартует в режиме исполнения скетча, или в скетче в настройках параметров отключен отладочный вывод в монитор порта

Cyxapuk11, 2024-01-03:
Перепрошил с дебагом (выключал когда ничего толком не работало т.к. думал что это именно на пины бесполезная по сути отладка). 
Вывод в порт заработал, а вместе с ним и успешно все подключилось в приложении с первой попытки. 
Извиняюсь за свои косяки и глупость, а вам огромное СПАСИБО за проявленное к идиоту терпение)

closed 2024-01-03 by Cyxapuk11