vvip-68/GyverPanelWiFi #301: ?

P55-rgb, 2024-03-01:
Вы на SD карте, что храните?
Ваша программа с карты выводит рисунки на светодиодную панель?

vvip-68, 2024-03-01:
Программа выводит с SD-карты рисунки на светодиодную панель.

P55-rgb, 2024-03-01:
Можете вашу программу пересобрать нано?
Мне на нано памяти не хватает, надо SD карту подключать.

vvip-68, 2024-03-01:
Нет, программа написана исключительно под микроконтроллеры ESP8266 / ESP32.
Попробуйте вместо Nano взять один из этих микроконтроллеров, в них памяти в разы больше чем на Nano.

closed 2024-03-01 by vvip-68

P55-rgb, 2024-03-01:
Вот такая плата?
https://aliexpress.ru/item/32556303666.html?sku_id=12000033912567280&spm=a2g2w.productlist.search_results.0.44193bf7dNAC33

или может у вас есть свои ссылки, на проверенные платы?

P55-rgb, 2024-03-01:
> Нет, программа написана исключительно под микроконтроллеры ESP8266 / ESP32. Попробуйте вместо Nano взять один из этих микроконтроллеров, в них памяти в разы больше чем на Nano.

На сколько светодиодов хватит этих мк с вашей программой?

vvip-68, 2024-03-01:
SD-карта с количеством светодиодов не связана никак. Вообще никак.
Эта карта что по ввашей ссылке - подойдет, но я бы предпочел такую: https://aliexpress.ru/item/32787418018.html
На них обоих именно для этой версии проекта - примерно 1024 светодиода. Для версии с Web-интерфейсом управления (соседний проект) - примерно 500-700.

Для вот такой платы https://aliexpress.ru/item/32807887667.html количество светодиодов при которых она еще будет работать - примерно 14-16 тыс. Но стоит иметь в виду, что чем больше диодов в цепочке, тем меньше скорость обновления всей цепочки диодов (всего изображения). На таком количестве скорость перерисовки может упасть до 2-4 кадров в секунду. Можно, конечно, увеличить скорость, если развести вывод на матрицу на параллельные сегменты.

P55-rgb, 2024-03-01:
У Гайвера программа позволяет считывать рис. с андроида и выводить его на экран.
Ваша программа это умеет? Если так, то SD карту можно не подключать?

При выключении питания ваш мк запоминает последнее состояние светодиодной панели? Т.е. если я его выключу, то он запомнит рисунок?

Я не нашел рис. или видео с вашей программы на андроид. Есть ли такое?
Я вашу программы установил, но без подключения просто экран с адресом и нет возможности увидеть возможности.

vvip-68, 2024-03-01:
1. Рисовать умеет. Загружать рисунки из картинок - нет.  Потому что "оцифровка" из картинки на низкое (как правило) разрешение матрицы - хрень полная.
2. Без SD-карты нарисовать картинку вы сможете, а сохранить для последующей загрузки на матрицу - нет. Точнее - почти нет. Оно может сохранять картинки и во внутреннюю память микроконтроллера (в его файловую систему), но ее объем достаточно невелик, много картинок не сохранишь. В зависимости от размеров матрицы - картинок 20-50, не больше.
3. Сохранять, как указано в п.2 можно. Но после выключения и включения рисунка на матрице не будет по понятным причинам - его нужно будет восстановить (загрузить) из сохраненного ранее в файл.
4. Идете в ютуб. Вбиваете в поиске vvip68 (или vvip-68). Вам предложат кучу роликов - люди собирали этот проект, снимали видео и выкладывали их в сеть.
5. Установленная программа без подключения к матрице (без визуализации) всё равно имеет возможность подключиться к ней с программы на андроиде (или с Web-страницы, если установлен скетч из соседнего репозитория - 1.14). Это позволит хотя бы посмотреть что из себя представляет программа управления матрицей.

vvip-68, 2024-03-01:
Поправка. Версия из этого репозитория - 1.13 довольно старая, я уже плохо помню. Но кажется она все-таки умеет загружать картинку из файла с картинкой (gif, png, bmp) на матрицу. Установите андроид-программу управления матрицей да посмотрите. Но качество такой "оцифровки" будет все равно так себе... Версия 1.14 точно не умеет. По указанным причинам.

P55-rgb, 2024-03-02:
Взял ваш проект, ни чего не менял, ошибка.
Как это вылечить?

[image]

FQBN: esp8266:esp8266:generic
Using board 'generic' from platform in folder: C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2
Using core 'esp8266' from platform in folder: C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2

"C:\\Users\\Peleng\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\Peleng\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2/tools/mkbuildoptglobals.py" "C:\\Progi\\arduino-ide_2.2.1_Windows_64bit\\resources\\app\\lib\\backend\\resources" 10607 "C:\\Users\\Peleng\\AppData\\Local\\Temp\\arduino\\sketches\\3DC3B6AC6EE496F2D89F31DCFC07E20C" "C:\\Users\\Peleng\\AppData\\Local\\Temp\\arduino\\sketches\\3DC3B6AC6EE496F2D89F31DCFC07E20C/core/build.opt" "C:\\Users\\Peleng\\Documents\\Arduino\\GyverPanelWiFi_v1.13/GyverPanelWiFi_v1.13.ino.globals.h" "C:\\Users\\Peleng\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2\\cores\\esp8266/CommonHFile.h"
default_encoding:       cp1251
Assume aggressive 'core.a' caching enabled.
Note: optional global include file 'C:\Users\Peleng\Documents\Arduino\GyverPanelWiFi_v1.13\GyverPanelWiFi_v1.13.ino.globals.h' does not exist.
  Read more at https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html
Detecting libraries used...
C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C/core/build.opt -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/libc/xtensa-lx106-elf/include -IC:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C/core -c @C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/warnings/none-cppflags -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DARDUINO_BOARD_ID="generic" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266 -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\generic C:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C\sketch\GyverPanelWiFi_v1.13.ino.cpp -o nul
Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]
ResolveLibrary(ESP8266WiFi.h)
  -> candidates: [ESP8266WiFi@1.0]
C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C/core/build.opt -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/libc/xtensa-lx106-elf/include -IC:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C/core -c @C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/warnings/none-cppflags -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DARDUINO_BOARD_ID="generic" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266 -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\generic -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src C:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C\sketch\GyverPanelWiFi_v1.13.ino.cpp -o nul
Alternatives for ESP8266mDNS.h: [ESP8266mDNS@1.2]
ResolveLibrary(ESP8266mDNS.h)
  -> candidates: [ESP8266mDNS@1.2]
C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\Peleng\AppData\Local\Temp\arduino\sketches\3DC3B6AC6EE496F2D89F31DCFC07E20C/core/build.opt -IC:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\User [... cut ...]
Используем библиотеку GyverButton в папке: C:\Users\Peleng\Documents\Arduino\libraries\GyverButton (устаревшее)
Используем библиотеку ESP Async E1.31 версии 1.0.3 из папки: C:\Users\Peleng\Documents\Arduino\libraries\ESP_Async_E1.31 
Используем библиотеку ESP Async UDP версии 1.1.0 из папки: C:\Users\Peleng\Documents\Arduino\libraries\ESPAsyncUDP 
Используем библиотеку LittleFS версии 0.1.0 из папки: C:\Users\Peleng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\LittleFS 
exit status 1

Compilation error: 'D4' was not declared in this scope

vvip-68, 2024-03-02:
Для начала, первым делом - хотя бы прочитать wiki проекта. Ну хотя бы раздел FAQ. Вторым шагом - прочитать комментарии в начале основного файла скетча.

Вы сделали неправильно абсолютно всё.
1. Выбрана "новейшая" версия IDE, которая имеет кучу глюков и несовместима с большинством проектов - там вечно вылазят какие-то непонятки. Начиная с некорректного отображения в области вывода, заканчивая неожиданными и непонятными вещами. Ставьте рекомендованную 1.8.19 
2. В проекте черным по белому написано, что проект рассчитан на версию ядра 2.7.4. С "новейшими" версиями ядра он или вообще не скомпилится или будет работать с глюками. Вы поставили последнюю из новейших.
3. В проекте ясно указано какие версии плат в настройках IDE. Вы выбрали первую попавшуюся generic/

То есть проигнорировав абсолютно все инструкции вы надеетесь получить какой-то вменяемый результат? Не получится.

P55-rgb, 2024-03-02:
Так верно?
[image]

vvip-68, 2024-03-02:
Разумеется нет. То есть, что описано в предыдущем посте вы просто решили проигнорировать?
Ну, успехов ковыряться дальше.

Раз уж вы решили остаться на ядре 3.1.2 - у вас же на скриншоте даже написано что где и как выбирать

[image]

P55-rgb, 2024-03-03:
Прошилось, теперь монитор СОМ порта выдает просто точки.
[image]
[image]

vvip-68, 2024-03-03:
Пытается подключиться к локальной сети. Если вы не настраивали параметры подключения - оно будет ждать 3 минуты или пока не нажмете кнопку для прерывания ожидания. Дальше создаст точку доступа и начнет работу.

P55-rgb, 2024-03-03:
[image]

Это сделал.
[image]

Наживаю Подключиться, пишет нет подключения. Что делать?
[image]

vvip-68, 2024-03-03:
На этот вопрос нет ответа. Никогда не сталкивался со случаем, чтобы телефон не мог подключиться к точке доступа. Возможно вам нужно скачать / прочитать инструкцию к вашему телефону, прочитать как подключаться с телефона к сетям и точкам доступа.

P55-rgb, 2024-03-03:
> На этот вопрос нет ответа. Никогда не сталкивался со случаем, чтобы телефон не мог подключиться к точке доступа. Возможно вам нужно скачать / прочитать инструкцию к вашему телефону, прочитать как подключаться с телефона к сетям и точкам доступа.

Сотовую сеть выкл. Подключился.

P55-rgb, 2024-03-03:
Не понятно, где настраивать матрицу.
В двух файлах или как?

[image]
[image]

vvip-68, 2024-03-03:
В зависимости от того, какие настройки указаны в файле a_def_hard выше.
А вообще - всё это настраивается из приложения.

P55-rgb, 2024-03-03:
> В зависимости от того, какие настройки указаны в файле a_def_hard выше. А вообще - всё это настраивается из приложения.

Это прочитал, не увидел, где менять размер матрицы.
https://github.com/vvip-68/GyverPanelWiFi/wiki/Настройка-параметров-в-приложении

Как загрузить на панель свой рисунок, через программу с андроида?

vvip-68, 2024-03-03:
Установить приложение WiFiPlayer на телефон, подключиться к матрице, перейти на вкладку "Картинки", нажать кнопку "Выбор", выбрать картинку. Затем - кнопку "Отправить" для отправки рисунка на матрицу

vvip-68, 2024-03-03:
[image]

vvip-68, 2024-03-03:
Размер матрицы меняется в приложении WiFiSetup, здесь
[image]

P55-rgb, 2024-03-03:
Матрицу подключил, в ручную рисует в программе Player. 
Загружаю рис. и программа закрывается. 
Она сама подрезает рис. или надо его готовить по пикселям?

vvip-68, 2024-03-03:
Если вы его (рисунок) подготовите - будет более точное соответствие желаемого тому, что получится в результате. В общем случае (насколько я помню) - рисунок просто трансформируется из исходного вида к пропорциям матрицы (вроде бы даже с искажением пропорций - не помню). Талее из загруженного рисунка кружочками (клеточками) выбирается область, вычисляется средний цвет этой области, который будет после отправлен на матрицу.

Если программа закрывается - возможно  предложенный рисунок слишком тяжелый и приложению на телефоне не хватает памяти. Другого предположения у меня нет.