vvip-68/GyverPanelWiFi #302: Нет возможности прописать доступ к Wi-Fi сети

khusamov, 2024-03-18:
Как попасть на эту страницу в приложении?
[image]

Дело в том, что у меня нет этих двух кнопок:
[image]

Остальные кнопки есть, даже больше чем на первом скриншоте. Но нужной кнопки нету. Как быть?

vvip-68, 2024-03-18:
Установить соответственно другое приложение для настройки, где эти кнопки есть.

khusamov, 2024-03-18:
@vvip-68 А другого же нету.
Я установил эту:

> Найдите в папке проекта Android/AppInventor файл приложения PanelSetupWiFi.apk и установите его на ваш смартфон. Запустите приложение.

khusamov, 2024-03-18:
@vvip-68 Спасибо! Понял где косяк))

closed 2024-03-18 by khusamov

vvip-68, 2024-03-18:
Ну вот в этом приложении эти кнопки и есть.
А что у вас в этом приложении?

> 
> 
>@vvip-68 А другого же нету.
>Я установил эту:
>>Найдите в папке проекта Android/AppInventor файл приложения PanelSetupWiFi.apk и установите его на ваш смартфон. Запустите приложение.
 
 
С уважением,
Вадим Пузанов

vvip-68, 2024-03-18:
Ну, ок