vvip-68/GyverPanelWiFi #307: Погода от Яндекс

Aleks000777999, 2024-11-03:
Здравствуйте. Не работает погода от яндекс. Установил последнюю версию WiFi-Panel-v1.13 released this Mar 4, 2023. Ткните носом где, что нужно исправить, прописать, или еще какие действия. Читал, что исправлено, но в этой версии не работает. Заранее спасибо.

vvip-68, 2024-11-03:
Исправлено. Подтверждаю. У всех, с кем я общался -  вроде работает....
Интересно про " released this Mar 4, 2023"... 
Последняя версия с исправлениями по яндексу выложена 31.10.2024...

Aleks000777999, 2024-11-03:
Извините, я не очень силен в этом. Дайте пожалуйста ссылку на последнюю версию с исправлениями.

vvip-68, 2024-11-03:
https://github.com/vvip-68/GyverPanelWiFi/wiki/FAQ
Первый же вопрос-ответ

Aleks000777999, 2024-11-03:
Спасибо. Пробую.

Aleks000777999, 2024-11-04:
ОГРОМНОЕ СПАСИБО. Работает.

closed 2024-11-04 by vvip-68

Aleks000777999, 2024-11-08:
Доброго времени. Побеспокою еще раз, с другим вопросом. Хотелось посмотреть как будет выглядеть анимация марио на моей матрице, размер 20х15. Но в списке не отображается марио и погода в приложении, знаю что нужна матрица 16х16. Но все же хотелось посмотреть. Был такой вопрос здесь, Вы отвечали человеку. Не могли бы Вы объяснить что именно нужно поправить в скетче во вкладке анимация, что бы появился весь список.  Вот Ваш скрин на тот вопрос
[image]

vvip-68, 2024-11-08:
Вот вообще не помню как там и что...
Обведенная рамкой конструкция говорит, что добавлять анимацию в список отображаемых нужно только когда ее размер (ширина, высота анимации) меньше или равна ширине и высоте матрицы соответственно...

Хотите посмотреть - отключите эту проверку вообще - ну то есть закомментируйте строки 120 и 125 что на этом скриншоте - тогда проверок вообще выполняться не будет - попадут все в список.

Aleks000777999, 2024-11-08:
Спасибо, я и имел ввиду убрать проверку. Вот еще бы знать как это сделать -(закомментируйте строки 120 и 125) ??? Извините я не программист...

vvip-68, 2024-11-08:
Можете вообще удалить эти две строки.
Но если попробовать, а потом вернут обратно - поставьте в начале каждой строки по две косых черты - //

Aleks000777999, 2024-11-08:
Спасибо пробую.

Aleks000777999, 2024-11-08:
Спасибо, получилось. УДАЧИ ВАМ !!!