vvip-68/GyverPanelWiFi #308: Ошибка при компиляции

Del3456, 2024-11-11:
exit status 1

Compilation error: 'uint8_t globalBrightness' redeclared as different kind of entity

ошибок в связи с этой затем много вываливается, могу лог файл приложить

vvip-68, 2024-11-11:
Да, тут кто-то совсем недавно писал про такое.
Я не понимаю, как вы этого добиваетесь.
Мне ни разу не удавалось такого получить.
Потом тот кто спрашивал про это - пропал, видимо как-то решил проблему.

Del3456, 2024-11-11:
да нашел соседнюю ветку, сейчас пробую разобраться, может дело в версии библиотек, буду пытаться откатиться, спасибо

Del3456, 2024-11-11:
установил версию 2.7.4 для ESP 8266. имею следующую ошибку:

C:\Users\del34\Documents\Arduino\GyverPanelWiFi_v1.13\effects2.ino: In function 'void fireworksRoutine()':
C:\Users\del34\Documents\Arduino\GyverPanelWiFi_v1.13\effects2.ino:76:312: error: 'LINEARBLEND_NOWRAP' was not declared in this scope
       if      (firework[i].the_figMode == 0) drawCircleBlend    (firework[i].the_figX, firework[i].the_figY, firework[i].the_figSize, ColorFromPalette( myRainbow_p, firework[i].the_figHue, (firework[i].the_figSize > firework[i].the_figSizeMax / 2) ? effectBrightness - effectBrightnessFaded : effectBrightness, LINEARBLEND_NOWRAP));
                                                                                                                                                                                                                                                                                                                        ^
C:\Users\del34\Documents\Arduino\GyverPanelWiFi_v1.13\effects2.ino:77:312: error: 'LINEARBLEND_NOWRAP' was not declared in this scope
       else if (firework[i].the_figMode == 1) drawSnowflakesBlend(firework[i].the_figX, firework[i].the_figY, firework[i].the_figSize, ColorFromPalette( myRainbow_p, firework[i].the_figHue, (firework[i].the_figSize > firework[i].the_figSizeMax / 2) ? effectBrightness - effectBrightnessFaded : effectBrightness, LINEARBLEND_NOWRAP));
                                                                                                                                                                                                                                                                                                                        ^

exit status 1

Compilation error: 'LINEARBLEND_NOWRAP' was not declared in this scope

vvip-68, 2024-11-11:
Две принципиальные ошибки. Прочитайте хотя бы комментарии в начале основного файла скетча.
Для работы скетчу нужно ядро версии 3.1.2 и FastLED версии 3.6.0

Del3456, 2024-11-11:
я прочитал комментарии, и для версии 1.13 указания следующие : Данный проект рассчитан на работу с версиями библиотек поддержки плат ESP версии 2.7.4 и библиотеки FastLED версии 3.4.0 или более новой - 3.5.0;

Я так и сделал откатился ESP с 3.1.2 до 2.7.4 и FastLed до 3.5

Del3456, 2024-11-11:
скачал сегодня, отсюда  из раздела code https://github.com/vvip-68/GyverPanelWiFi версия 1.13

vvip-68, 2024-11-11:
Вот что я вижу по указанной вами ссылке:
[image]

Где здесь про версию ядра 2.7.4 и FastLED 3.5.0?
Очистите кэш браузера, если показывает и загружает что-то другое...

Del3456, 2024-11-11:
спасибо, буду пробовать

Del3456, 2024-11-11:
однако, вот здесь https://github.com/vvip-68/GyverPanelWiFi внизу страницы стоит поменять тогда текст, я изначально делал по нему
[image]

vvip-68, 2024-11-11:
Хм... Действительнго... Хотя, помнится я это исправлял... Куда пропало...
Поправлю.

vvip-68, 2025-12-07:
Какая именно ошибка возникает?