vvip-68/GyverPanelWiFi #309: effecs2.ino содержит необъявленную переменную

csandr, 2024-11-12:
effecs2.ino содержит необъявленную переменную LINEARBLEND_NOWRAP. Проект по умолчанию не компилируется.

Compilation error: 'LINEARBLEND_NOWRAP' was not declared in this scope

https://github.com/vvip-68/GyverPanelWiFi/blob/5b8067b3d03aa4ed02c65e6ac0fdf1807851a05f/firmware/GyverPanelWiFi_v1.13/effects2.ino#L76C312-L77

vvip-68, 2024-11-12:
Внимательно прочитайте инструкцию и установить ПРАВИЛЬНЫЕ версии ядра и библиотек.

csandr, 2024-11-12:
Да, понял это. Иду по пунктам. Про версии пока ничего не было. Пошёл искать...

Спасибо

csandr, 2024-11-12:
https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%9F%D0%BE%D1%88%D0%B0%D0%B3%D0%BE%D0%B2%D0%B0%D1%8F-%D0%B8%D0%BD%D1%81%D1%82%D1%80%D1%83%D0%BA%D1%86%D0%B8%D1%8F-%D0%BF%D0%BE%D0%B4%D0%B3%D0%BE%D1%82%D0%BE%D0%B2%D0%BA%D0%B8-%D1%81%D1%80%D0%B5%D0%B4%D1%8B-%D0%B4%D0%BB%D1%8F-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B0#%D1%88%D0%B0%D0%B3-6

Установить есть, а версии нет. Ищу...

csandr, 2024-11-12:
Нашёл:
https://github.com/vvip-68/GyverPanelWiFi#%D0%B2%D0%B0%D0%B6%D0%BD%D0%BE
ESP8266 версии 3.2.1
Крайняя доступная на текущий момент 3.1.2.
В файле указано верно
https://github.com/vvip-68/GyverPanelWiFi/blob/5b8067b3d03aa4ed02c65e6ac0fdf1807851a05f/firmware/GyverPanelWiFi_v1.13/GyverPanelWiFi_v1.13.ino#L28

vvip-68, 2024-11-12:
Да хоть на самом начальном экране проекта на гитхабе (внизу под "Важно"), хоть в FAQ хоть в основном файле проекта вверху целая портянка комментов как и что ожидается.

Даа, ядро 3.1.2, FastLED 3.6.0

closed 2024-11-12 by csandr