vvip-68/GyverPanelWiFi #312: Wifi ч

kost111233, 2024-12-01:
Добрый вечер. Извините за глупые и повторяющиеся вопросы, поискал в закрытых вопросах, не нашел ответа.

данный проект собирал года 3 назад и каждый год просто запускаю и все работает. Но летом сменил роутер и адрес поменялся с 192.168.31.. на 192.168.50… При запуске гирлянда работает на старых настройках, меняет режимы, но управлять ей не получается. Нашел способ при запуске нажать кнопку, что бы к ней подключиться через приложение с адресом 192.168.4.1, но ничего не создается. Как надо жать эту кнопку при запуске и как долго?

vvip-68, 2024-12-01:
Сразу при старте, ну, секунд 5...
Или просто выключите роутер на время, перезапустите гирлянду и ждите минуты три...
Или смотрите в монитор порта, подключив гирлянду - когда там будет надпись о попытке подключения к сети и предложение нажать кнопку для создания точки доступа - вот и нажимайте в это время...

kost111233, 2024-12-01:
> Сразу при старте, ну, секунд 5...
> Или просто выключите роутер на время, перезапустите гирлянду и ждите минуты три...
> Или смотрите в монитор порта, подключив гирлянду - когда там будет надпись о [...]

[image]

выключил роутер и появился вайфай, но подключиться к нему не получается. Вводил пароль 12341234

vvip-68, 2024-12-01:
Ну, тут ничего не подскажу. Пароль по умолчанию - именно такой. Может вы его меняли где / когда?... Вспоминайте.

kost111233, 2024-12-01:
Вроде не менял. Заново не осилю прошить) прошлый раз душу продал что бы нужные библиотеки поставить и прошить) так в целом она работает, просто не запускаются приложения. Жаль что она ип адрес не меняет сама, вайфай то прежний остался. Да и в роутере не хочется его менять, так как двум другим прописал статичный ип с 192.168.50…

kost111233, 2024-12-01:
> Ну, тут ничего не подскажу. Пароль по умолчанию - именно такой. Может вы его меняли где / когда?... Вспоминайте.

Еще у вас в гайдах написано что должна точка PanelAP создаться, а у меня MatrixAP создалась. Все правильно?

vvip-68, 2024-12-01:
Ну, может, в каких-то очень древних версиях и так называлась.
Но это очень-очень давно было...
В относительно свежих версиях (даже двухлетней давности) - она точно называется PanelAP...

У вас, видимо, совсем древняя версия прошивки стоит...
Прошить на свежую в общем то совсем не сложно...

Для версии 1.13 - скачать с репозитория https://github.com/vvip-68/GyverPanelWiFi
Для версии 1.14 - из https://github.com/vvip-68/LedPanelWiFi

В ардуино IDE библиотеки скопировать из папки Libraries
FastLED поставить из менеджера библиотек версии 3.6.0
Из менеджера плат версию ядря для 8266 - уставить 3.1.2? если esp32 - 2.0.12

Поменять размеры на свои, залить прошивку
Если 1.14 - залить файловую систему на МК - там WEB-странички для управления из браузера
Делов то...

kost111233, 2024-12-01:
Решил проблему. Зашел в настройки роутера, изменил его ип 192.168.50.1 на старый 192.168.31.1 , далее смог подключиться к матрице и изменить там ип. Снова в роутере вернул как было, зашел на матрицу и пришлось еще раз новый ип добавить. Теперь все подключается и снова работкет. Спасибо за помощь и за крутой проект! С наступающим!

closed 2024-12-06 by vvip-68