vvip-68/GyverPanelWiFi #318: Ошибка компиляции на ESP 32 2.0.17

CrazyGhostRider, 2024-12-19:
Arduino IDE - v1.8.19
WiFiPanel v.1.13.2024.1212
ESP32   - 2.0.17
FastLED      - 3.7.0
Все остальные библиотеки закинул

Решил обновиться с прошлого года. Но получил ошибку:
[2024_12_16 16-47-05.txt](https://github.com/user-attachments/files/18199482/2024_12_16.16-47-05.txt)

Про ESP32 - 2.0.17 написано в строке 29 файла "GyverPanelWiFi_v1.13.ino"
На ESP32 - 2.0.14 всё скомпилилось.

И ещё заметил, что раньше гирлянда не загоралась при включении платы, если задано время работы. (Например: 21:00 - демо режим; 23:00 - выключить. А включил всё в 18:00) А сейчас сразу включается демо режим.

vvip-68, 2024-12-19:
Ну скомпилилось - и хорошо. У меня и на 2.0.14 и на 2.0.17 собирается без проблем.
Особо мне кажется не принципиально. По логам - похоже на конфликт каких-то библиотек.
Я бы посоветовал удалить абсолютно все и установить их из проекта.

Про включение - тут уже лет 5 ничего не менялось. Состояние гирлянды сохраняется в EEPROM.
При подаче питания на контроллер - все настройки (в т.ч и статус включено / выключено) загружается из сохраненных в постоянной памяти и применяются.

Настройки по времени - не анализируются вперед/назад. 
В каждой итерации цикла loop() проверяется текущее время и нет ли чего либо назначенного на это время. Если есть - выполняется.

CrazyGhostRider, 2024-12-19:
Спасибо.
Кстати, на Arduino IDE 2.3.4 есть смысл переходить? И мониторинг порта нормально работает?

vvip-68, 2024-12-19:
Ой, была б моя воля - я б вообще на 1.8.19 оставался...

closed 2024-12-19 by CrazyGhostRider